arduino/arduino-cli #1550: Inconsistency in how project subfolders are compiled

snives, 2021-11-17:
I've searched for answers but didn't see this topic, apologies if its been discussed previously, please provide a link to that issue if so.

## Bug Report
Inconsistency in how project specific libraries are compiled.  The name of the library subfolder actually changes the behavior of the compiled executable.  This seems arbitrary and relies upon magic names for projects to work.  User will receive undefined reference errors during linking if including project subfolder not named SRC.

My project consists of a single .ino file in the root of the project folder as well as several .h and .cpp files in a subfolder which constitutes a project specific library.  In the .ino the .h files are referenced with an include statement.

Tested with Arduino IDE v1.8.15.

### Current behavior

When using the `#include "subfolder/somefile.h"` declaration to include a header it will not compile the corresponding somefile.cpp implementation file.  Weirdly, it does copy "subfolder/somefile.cpp" to the temp folder for compiling the sketch.  But any references to static variables defined in the implementation will fail resolving at link time and will show an error for each reference of that variable  `undefined reference to SomeFunction()`.  To me this shows its not compiling the somefile.cpp file.

However, If the subfolder is renamed to the magic folder name "Src" and recompiled then the implementation file `somefile.cpp` in that folder WILL get compiled and linked.  And the sketch will function as intended.

### Expected behavior

I would have expected that the Arduino compiler would compile the project consistently regardless of the subfolder's (or libraries') name.  It appears that "src" is a magic folder name that is required to compile the project successfully.

This becomes an issue if someone is writing multiple project specific libraries and they cannot both be named `Src`.

### Environment

- Arduino IDE v1.8.15
- Windows 10 x64

### Additional context

If this is the desired functionality can you please provide a link to the documentation which states this is the intended behavior?  That would be much appreciated.  Thank you

per1234, 2021-11-17:
Hello @snives. Thanks for taking the time to submit an issue.

>can you please provide a link to the documentation which states this is the intended behavior?

Here: it is in the Arduino sketch specification:
https://arduino.github.io/arduino-cli/latest/sketch-specification/#src-subfolder

>If the subfolder is renamed to the magic folder name "Src"

Please note that it should be named `src` (all lower case).

>This becomes an issue if someone is writing multiple project specific libraries and they cannot both be named Src.

The contents of the `src` subfolder are compiled recursively, so you can put as many subfolders as you like under it, separating the libraries into any folder structure that is needed.

Since this is the expected behavior and is fully documented, I'll go ahead and close this as resolved.

closed 2021-11-17 by per1234

snives, 2021-11-17:
Thank you, that was helpful.