arduino/arduino-cli #1562: monitor do apply serial port configuration change like baudrate

romainreignier, 2021-11-23:
## Bug Report

### Current behavior

With the new 0.20.0 version, the `monitor` has been added but the configuration changes are not applied.

For example, changing the baudrate to 115200 is not applied with this command:

```
arduino-cli monitor -p /dev/ttyUSB0 -c baudrate=115200
```

The configuration is recognized because the following output is given:

```
Monitor port settings:
baudrate=115200
Connected to /dev/ttyUSB0! Press CTRL-C to exit.
```

And the output of the serial port is not readable, showing not printable characters because of the bad baudrate settings.

### Expected behavior

I would expect to see the strings printed by my Arduino board at 115200 bauds.

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli  Version: 0.20.0 Commit: 553c6375 Date: 2021-11-23T14:35:08Z` and git master version: `arduino-cli  Version: git-snapshot Commit: c756a0f1 Date: 2021-11-23T17:20:00Z`
- OS and platform: Linux, Ubuntu 18.04.6

### Additional context

After ading some logs from the source code, I did not found anywhere where the serial port config where sent to the `serial-monitor` program.

Adding this quick and dirty patch fixed the issue:

```patch
diff --git a/commands/monitor/monitor.go b/commands/monitor/monitor.go
index 8c217f81..49cbab66 100644
--- a/commands/monitor/monitor.go
+++ b/commands/monitor/monitor.go
@@ -84,6 +84,10 @@ func Monitor(ctx context.Context, req *rpc.MonitorRequest) (*PortProxy, *pluggab
                return nil, nil, &arduino.FailedMonitorError{Cause: err}
        }

+       for _, setting := range req.GetPortConfiguration().Settings {
+               m.Configure(setting.SettingId, setting.Value)
+       }
+
        return &PortProxy{
                rw:               monIO,
                changeSettingsCB: m.Configure,
```

romainreignier, 2021-11-23:
As a side note, an example of changing the baudrate should be added to the documentation.

cmaglie, 2021-12-23:
Thanks @romainreignier!

PR #1604 (2021-12-23): Bugfix: apply monitor settings on monitor opening
**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] `UPGRADING.md` has been updated with a migration guide (for breaking changes) [... cut]
commit 12adc538 (2021-12-23): Bugfix: apply monitor settings on monitor opening (#1604)

Fix #1562
closed 2021-12-23 by cmaglie