arduino/arduino-cli #1564: `monitorDependencies` are not installed with platform

per1234, 2021-11-23:
From https://arduino.github.io/arduino-cli/dev/package_index_json-specification/#platforms-definitions

> - `monitorDependencies`: the Pluggable Monitors needed by this platform. Each monitor is referenced by the pair
>   (`packager`, `name`), the `version` is not specified because the latest installed monitor tool will always be used.
>   Like `toolsDependencies` they will be installed by Boards Manager along with the platform and can reference tools
>   available in other packages as well, even if no platform of that package is installed.

However, I find that the tool is not installed.

## Steps to reproduce

Create a package index that contains a platform with a tool reference in its `monitorDependencies` field. For example:

<details>
<summary>Click to expand</summary>

```json
{
  "packages": [
    {
      "name": "per1234",
      "maintainer": "foo",
      "websiteURL": "http://example.com/",
      "email": "foo@example.com",
      "help": {
        "online": "http://example.com"
      },
      "platforms": [
        {
          "name": "Foo Boards",
          "architecture": "avr",
          "version": "1.2.3",
          "category": "Contributed",
          "help": {
            "online": "http://example.com/"
          },
          "url": "http://downloads.arduino.cc/cores/avr-1.8.3.tar.bz2",
          "archiveFileName": "avr-1.8.3.tar.bz2",
          "checksum": "SHA-256:de8a9b982477762d3d3e52fc2b682cdd8ff194dc3f1d46f4debdea6a01b33c14",
          "size": "4941548",
          "boards": [],
          "toolsDependencies": [],
          "monitorDependencies": [
            {
              "packager": "per1234",
              "name": "some-tool"
            }
          ]
        }
      ],
      "tools": [
        {
          "name": "some-tool",
          "version": "0.42.0",
          "systems": [
            {
              "host": "i686-pc-linux-gnu",
              "archiveFileName": "serial-discovery_v1.3.0-rc1_Linux_32bit.tar.bz2",
              "url": "https://downloads.arduino.cc/discovery/serial-discovery/serial-discovery_v1.3.0-rc1_Linux_32bit.tar.gz",
              "size": 1633143,
              "checksum": "SHA-256:2fb17882018f3eefeaf933673cbc42cea83ce739503880ccc7f9cf521de0e513"
            },
            {
              "host": "x86_64-pc-linux-gnu",
              "archiveFileName": "serial-discovery_v1.3.0-rc1_Linux_64bit.tar.bz2",
              "url": "https://downloads.arduino.cc/discovery/serial-discovery/serial-discovery_v1.3.0-rc1_Linux_64bit.tar.gz",
              "size": 1688362,
              "checksum": "SHA-256:e0e55ea9c5e05f12af5d89dc3a69d63e12211f54122b4bf45a7cab9f0a6f89e5"
            },
            {
              "host": "i686-mingw32",
              "archiveFileName": "serial-discovery_v1.3.0-rc1_Windows_32bit.zip",
              "url": "https://downloads.arduino.cc/discovery/serial-discovery/serial-discovery_v1.3.0-rc1_Windows_32bit.zip",
              "size": 1742668,
              "checksum": "SHA-256:4acfe521d6fc3b29643ab69ced246d7dd20637772fc79fc3e509829c18290d90"
            },
            {
              "host": "x86_64-mingw32",
              "archiveFileName": "serial-discovery_v1.3.0-rc1_Windows_64bit.zip",
              "url": "https://downloads.arduino.cc/discovery/serial-discovery/serial-discovery_v1.3.0-rc1_Windows_64bit.zip",
              "size": 1709333,
              "checksum": "SHA-256:82b2edea04f7c97b98cbb04de95ec48be95de64fa5f196d730dc824d7558b952"
            },
            {
              "host": "x86_64-apple-darwin",
              "archiveFileName": "serial-discovery_v1.3.0-rc1_macOS_64bit.tar.bz2",
              "url": "https://downloads.arduino.cc/discovery/serial-discovery/serial-discovery_v1.3.0-rc1_macOS_64bit.tar.gz",
              "size": 964596,
              "checksum": "SHA-256:ec4be0f5c1ed6af3f31bb01ed6a5433274a76a1dc7cb68d39813b2b0475d7337"
            },
            {
              "host": "arm-linux-gnueabihf",
              "archiveFileName": "serial-discovery_v1.3.0-rc1_Linux_ARMv6.tar.bz2",
              "url": "https://downloads.arduino.cc/discovery/serial-discovery/serial-discovery_v1.3.0-rc1_Linux_ARMv6.tar.gz",
              "size": 1570847,
              "checksum": "SHA-256:9341e2541ad41ee2cdaad1e8d851254c8bce63c937cdafd57db7d1439d8ced59"
            },
            {
              "host": "arm64-linux-gnueabihf",
              "archiveFileName": "serial-discovery_v1.3.0-rc1_Linux_ARM64.tar.bz2",
              "url": "https://downloads.arduino.cc/discovery/serial-discovery/serial-discovery_v1.3.0-rc1_Linux_ARM64.tar.gz",
              "size": 1580108,
              "checksum": "SHA-256:1da38f94be8db69bbe26d6a95692b665f6bc9bf89aa62b58d4e4cfb0f7fd8733"
            }
          ]
        }
      ]
    }
  ]
}
```

</details>

Install the platform and then check to see whether the tool was installed.

### Demonstration

```
$ arduino-cli version
arduino-cli alpha Version: nightly-20211114 Commit: bf4a784 Date: 2021-11-14T01:27:21Z

$ arduino-cli core update-index --additional-urls file:///tmp/package_per1234_index.json
Updating index: package_index.json downloaded
Updating index: package_index.json.sig downloaded
Updating index: package_per1234_index.json downloaded

$ arduino-cli core install --additional-urls file:///tmp/package_per1234_index.json per1234:avr
Downloading packages...
per1234:avr@1.2.3 already downloaded
Installing platform per1234:avr@1.2.3...
Configuring platform....
Platform per1234:avr@1.2.3 installed

$ ls ~/.arduino15/packages/per1234
hardware
```

## Environment

- OS: Ubuntu 20.04.3
- Arduino CLI: nightly-20211114 Commit: bf4a784 Date: 2021-11-14T01:27:21Z

## Additional information

Tool installation works just as expected for `toolsDependencies` and `discoveryDependencies`.

PaulStoffregen, 2022-01-20:
I am also experiencing this issue while attempting to create a plugable monitor for Teensy.

This is my package index file:
https://www.pjrc.com/teensy/td_156/package_teensy_index.json

When I run "./arduino-cli core install teensy:avr", this is the result (no teensy-discovery tool installed)

```
Downloading packages...
teensy:teensy-tools@1.56 already downloaded
teensy:teensy-compile@1.56 already downloaded
teensy:teensy-discovery@1.56 already downloaded
teensy:avr@1.56 already downloaded
Installing teensy:teensy-tools@1.56...
teensy:teensy-tools@1.56 installed
Installing teensy:teensy-compile@1.56...
teensy:teensy-compile@1.56 installed
Installing teensy:teensy-discovery@1.56...
teensy:teensy-discovery@1.56 installed
Installing platform teensy:avr@1.56...
Configuring platform....
Platform teensy:avr@1.56 installed
```

I tested with this arduino-cli version (nightly build)

    arduino-cli  Version: nightly-20220120 Commit: 5beeba4 Date: 2022-01-20T01:32:18Z

PaulStoffregen, 2022-01-23:
Quick followup.  Specifying my pluggable monitor in toolsDependencies seems to work.  Even though it wasn't specified by monitorDependencies, running "arduino-cli monitor -v --port usb16/16-1/16-1.2" does seem to cause it to run, when I have the proper line in platform.txt.

Of course I still need to do all the work to implement the new protocol with localhost socket....

closed 2022-01-25 by cmaglie