arduino/arduino-cli #1568: arduino-cli core install arduino:avr not working on low memory system

evanharwin, 2021-11-24:
## Bug Report

### Current behavior

Trying to install `arduino:avr` support, as I belive this is neccessary for my Arduino UNO.

```bash
arduino-cli core install arduino:avr --verbose
```

Process is killed with out of memory error. Logs show it was on installation of `arduino:avrdude@6.3.0-arduino17`:

```
INFO[0000] Using config file: /home/pi/.arduino15/arduino-cli.yaml
INFO[0000] arduino-cli version 0.20.0
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Adding libraries dir                          dir=/home/pi/Arduino/libraries location=user
INFO[0001] Checking signature                            index=/home/pi/.arduino15/package_index.json signatureFile=/home/pi/.arduino15/package_index.json.sig trusted=true
INFO[0001] Loading hardware from: /home/pi/.arduino15/packages
INFO[0001] Loading package builtin from: /home/pi/.arduino15/packages/builtin
INFO[0001] Checking existence of 'tools' path: /home/pi/.arduino15/packages/builtin/tools
INFO[0001] Loading tools from dir: /home/pi/.arduino15/packages/builtin/tools
INFO[0001] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0001] Loaded tool                                   tool="builtin:mdns-discovery@1.0.1"
INFO[0001] Loaded tool                                   tool="builtin:serial-discovery@1.3.0-rc1"
INFO[0001] Loaded tool                                   tool="builtin:serial-monitor@0.9.1"
INFO[0043] Executing `arduino-cli core install`
INFO[0043] Running from console, will run post-install by default
Downloading packages...
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 downloaded
arduino:avrdude@6.3.0-arduino17 downloaded
arduino:arduinoOTA@1.3.0 downloaded
arduino:avr@1.8.3 downloaded
INFO[0139] Installing tool                               Tool="arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7"
Installing arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7...
INFO[2047] Tool installed                                Tool="arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7"
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 installed
INFO[2047] Installing tool                               Tool="arduino:avrdude@6.3.0-arduino17"
Installing arduino:avrdude@6.3.0-arduino17...
Killed
```

And kernel logs show it was an out of memory error:
```
Out of memory: Killed process 729 (arduino-cli) total-vm:865672kB
```

### Expected behavior

Installation to complete successfully.

### Environment

- CLI version (output of `arduino-cli version`): "arduino-cli  Version: 0.20.0 Commit: 553c6375 Date: 2021-11-23T14:35:08Z"
- OS and platform: "Raspbian GNU/Linux 10 (buster)" on "Raspberry Pi Model B"

umbynos, 2023-02-02:
Should be solved in https://github.com/codeclysm/extract/pull/10
Could you please verify?

cmaglie, 2023-03-17:
Closing as solved, feel free to reopen (or open a new issue9 if the problem persists.

closed 2023-03-17 by cmaglie