arduino/arduino-cli #1587: Trying to kill a discovery process on Windows might fail with a `TerminateProcess access is denied`

silvanocerza, 2021-12-10:
## Bug Report

### Current behavior

I only verified this behaviour when running the `arduino-cli` in `daemon` mode. 
It probably happens also when running a CLI command but it should not affect the end user in that case since the discovery processes would quit in any case. 

[This line](https://github.com/arduino/arduino-cli/blob/d458040fea6f8d6148acad1bc0e3a0e2e8d8b27d/arduino/discovery/discovery.go#L269) returns an `error` so the CLI doesn't consider the discovery process to be closed. 

This prevents the `quit` message to be sent to any client listening for boards connection and disconnection event, so these client won't know they must restart the board list watch process and listen from a new communication channel, thus the clients won't see any more events.

The `quit` message has been added with #1460.

### Expected behavior

### Environment

- CLI version (output of `arduino-cli version`): `0.20.2`
- OS and platform: Windows 

### Additional context

My guess is that it's failing because the process we're trying to kill already has quit, we should probably only kill it if it's still running.

cmaglie, 2023-03-17:
@MatteoPologruto could you check if this is still an issue?

MatteoPologruto, 2023-04-05:
We were not able to reproduce this error. If this kind of problem will present itself in the future, we will open it again. In this event, could you provide us with the steps to reproduce it @silvanocerza ?

closed 2023-04-05 by MatteoPologruto

silvanocerza, 2023-04-06:
@MatteoPologruto no idea really, if I remember correctly it was happening while restarting the discovery processes or something like that.

I might be wrong though, the issue is quite old. 😄