arduino/arduino-cli #1590: Architecture or board not supported with ESP32 and SD Library

danilopinotti, 2021-12-15:
**Describe the bug**
When I try to compile some ESP32 code that uses SD library, the following message appears:
```
/home/danilopinotti/Arduino/libraries/SD/src/utility/Sd2PinMap.h:524:2: error: #error Architecture or board not supported.
 #error Architecture or board not supported.
```
I think this is a bug because the **exact** same code compiles (and works) within Arduino IDE 1.18.

I think it is using the default Arduino SD library instead SD library of Espressif that was installed with the board via board manager (https://github.com/espressif/arduino-esp32/tree/master/libraries).

**To Reproduce**
Steps to reproduce the behavior:

**Import ESP32 boards:**
1. Go to: File -> Preferences -> Aditional boards manager URLs
2. Insert `https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json` and click at OK button.
3. Tools -> Board -> Boards Manager
4. Search by "esp32" and install version **1.0.6**

**Select Board**
1. Go to Tools -> Board -> esp32
2. Select "ESP32-Dev-Module"

**Code**
Use this simple sketch and try to compile
```ino
#include "SD.h"

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

**Expected behavior**
The code should compile

**Desktop (please complete the following information):**
 - OS: Linux - PopOS 20.04 LTS
 - Version: 2.0.0-rc2

**Additional context**
In some tests, using v2.0 of esp boards, the code works.

per1234, 2021-12-15:
Hi @danilopinotti. Thanks for taking the time to submit an issue.

The reason for this behavior is explained in this response to a previous report about the same thing: https://github.com/arduino/arduino-cli/issues/1548#issuecomment-968085608

So this is the expected behavior and has been fixed in the latest version of the ESP32 boards platform. Just use Boards Manager to update to version 2.0.1 and the sketch will compile as expected.

The reason Arduino IDE 1.8.x behaves differently is because it is using an older version of Arduino CLI from before the "Library Name Priority" feature was introduced.

closed 2021-12-15 by per1234

mathemaphysics, 2024-07-05:
Same problem in PlatformIO. A have to comment out the pointless #error to make it work.

I'm already at the latest update. I don't understand how I'm supposed to update Arduino in PlatformIO when it's already up to date.