arduino/arduino-cli #1605: "arduino-cli upload" ignores port setting in "sketch.json"

ShyLionTjmn, 2021-12-24:
i have sketch.json:

`{
  "cpu": {
    "fqbn": "arduino:avr:pro",
    "port": "/dev/ttyUSB0"
  }
}`

Still, every time i run "arduino-cli upload" i have to add "-p /dev/ttyUSB0" option.
Am i doing something wrong?

davegarthsimpson, 2022-11-14:
Hi @per1234 are there any further actions on this?

per1234, 2022-11-15:
@davegarthsimpson the problem is that the previous `sketch.json` system requires a URI style protocol+port format like `serial:///dev/ttyUSB0`. @ShyLionTjmn must have manually added the unsupported port format into the `cpu.port` field of the file, since if `board attach --port` is used, the port has the correct format. Unfortunately, Arduino CLI does not communicate the problem to the user.

This problem will be resolved by the pending complete reworking of the `board attach` command in https://github.com/arduino/arduino-cli/pull/1930, which adds support for the more familiar `/dev/ttyUSB0` port format supported by all Arduino CLI's `--port` flags. That change also means the port is passed directly to the upload tool rather than parsing it for a protocol (there is a dedicated flag and field for the protocol), so the user would likely receive a more meaningful error from the upload tool in the event they specify a port of an invalid format in the file.

closed 2022-11-21 by cmaglie