arduino/arduino-cli #1607: Q: how can one pass `ConfigOption`s for the compilation?

4ntoine, 2021-12-28:
## Bug Report

### Background

Boards can [provide](https://github.com/arduino/arduino-cli/blob/master/rpc/cc/arduino/cli/commands/v1/board.proto#L56) `ConfigOption`s describing the board. For instance, "Nano" (arduino:avr:nano) provides 3 options (328, 328old and 168) based on the cpu and bootloader. Some arguments might make sense for the compilation, eg. cpu frequency.

### Current behavior

There is no config option argument to pass (or it's unclear to me on how to do it).
There are some ["custom build properties"](https://github.com/arduino/arduino-cli/blob/e63c798cda729592f449660cbb0d2a8dc9473ca0/rpc/cc/arduino/cli/commands/v1/compile.proto#L46), but it's unclear if it's what i mean.

### Expected behavior

There is a way to pass a `ConfigOption` value and it's documented.

### Environment

- CLI version (output of `arduino-cli version`): 0.20.2
- OS and platform: any

### Additional context

per1234, 2021-12-28:
Hi @4ntoine. These custom board option are specified as part of the FQBN.

This is documented in the platform specification:

https://arduino.github.io/arduino-cli/dev/platform-specification/#custom-board-options

and the FAQ:

https://arduino.github.io/arduino-cli/dev/FAQ/#how-to-set-multiple-board-options

So if you wanted to specify [the `atmega168` option ID](https://github.com/arduino/ArduinoCore-avr/blob/1.8.4/boards.txt#L207) for the `arduino:avr:nano` board's `processor` menu ID, the FQBN would be:

```
arduino:avr:nano:processor=atmega168
```

You can specify values for as many menu IDs as you like via a comma-separated list.

We are using this issue to track need to communicate better about what FQBN is and how it works: https://github.com/arduino/arduino-cli/issues/355

closed 2021-12-28 by per1234

4ntoine, 2021-12-28:
@per1234 Thank you