arduino/arduino-cli #1609: `arduino-cli board list --format json` command doesn't return `VID` and `PID` with the `--watch` flag.

abdullahmahboob, 2021-12-31:
Hey, this PR mentions that the `arduino-cli board list --format json` command returns `VID` and `PID` with the `--watch` flag as well, but I can't make it to return `VID` and `PID`. Is there some other flag for it? Here's how it works for me:

https://user-images.githubusercontent.com/26253942/147410728-c0f7a07f-f930-449d-ba87-0d001bf6e375.mov

_Originally posted by @abdullahmahboob in https://github.com/arduino/arduino-cli/issues/1076#issuecomment-1001188741_

silvanocerza, 2022-01-03:
You're right, the output from `board list --format json` and `board list --watch --format json` should be identical but it's not.
Thanks for the report, we'll fix this. :pray:

per1234, 2022-12-16:
Fixed by #1845

closed 2022-12-16 by per1234