arduino/arduino-cli #1611: Q: how to compile with library used?

4ntoine, 2022-01-01:
## Bug Report

### Current behavior

Sketch compilation with library used succeeds over cmd, but fails over grpc.

cmd:
```
> arduino-cli sketch new serv
> cat /tmp/serv/serv.ino 
#include <Servo.h>

Servo myservo;

void setup() {
}

void loop() {
}
> arduino-cli core list
ID          Installed Latest Name              
arduino:avr 1.8.3     1.8.4  Arduino AVR Boards
> arduino-cli compile --fqbn arduino:avr:uno /tmp/serv
Sketch uses 1154 bytes (3%) of program storage space. Maximum is 32256 bytes.
Global variables use 50 bytes (2%) of dynamic memory, leaving 1998 bytes for local variables. Maximum is 2048 bytes.
```

grpc (dart code):
```
// platform is previously installed as follows:
var responseStream = client.platformInstall(PlatformInstallRequest(
        instance: cliInstance,
        platformPackage: platform.getPackage(),
        architecture: platform.getArchitecture(),
        version: platform.availableVersion
    ));
...
// create
final createResponse = await client.create(CreateRequest.getDefault());
// update index
await for (var res in client.updateIndex(UpdateIndexRequest(
        instance: createResponse.instance))) {
        // nothing
      }
// update libs
await for (var res in client.updateLibrariesIndex(UpdateLibrariesIndexRequest(
          instance: createResponse.instance))) {
        // nothing
      }
// init
await for (var res in client.init(InitRequest(
          instance: createResponse.instance))) {
        // nothing
      }
// compile
var sources = Map<String, String>();
    this.sketchPresenter!.model.files.forEach((sourceFile) {
      sources[sourceFile.name] = sourceFile.text!;
    });
final compileStream = client.compile(CompileRequest(
        instance: cliInstance,
        sketchPath: someSketchPath,
        fqbn: "arduino:avr:uno",
        verbose: true,
        sourceOverride: sources,
        buildPath: someBuildPath
    )).asBroadcastStream();
```

Getting:
```
Alternatives for Servo.h []
ResolveLibrary(Servo.h)
-> candidates: []
/user/ser/ser.ino:1:10: fatal error: Servo.h: No such file or directory
...
```

### Expected behavior

Compilation over gRPC succeeds.

### Environment

- CLI version (output of `arduino-cli version`): `0.20.2`
- OS and platform: macos

### Additional context

4ntoine, 2022-01-01:
PS. cmd and grpc tested on different devices (probably makes sense)

4ntoine, 2022-01-01:
Ah, cmd executing on the same device failed - it seems that "Servo" is not installed by `platform install` command. Is it expected to be installed additionally?

4ntoine, 2022-01-01:
Yeah, i had to install it additionally and it works now. The only thing that is not clear what's the default llibs list to be installed? Is it an expected behaviour?

closed 2022-01-01 by 4ntoine

per1234, 2022-01-01:
>what's the default llibs list to be installed?

Hi @4ntoine. You can learn about the "platform bundled libraries" that are installed along with an Arduino platform here:

https://arduino.github.io/arduino-cli/0.20/platform-specification/#platform-bundled-libraries

Although it is a common convention to provide the fundamental SPI and Wire libraries at least, the platform author is free to choose the libraries that are bundled, so there is no fixed list of libraries to be installed with a platform.

>Is it an expected behaviour?

Yes. Unlike the Arduino IDE, the installation of Arduino CLI itself doesn't come with any libraries. So, other than the platform bundled libraries of the board you are compiling for, all others must be installed via `lib install`. There is currently an odd exception to that, which is described here: https://github.com/arduino/arduino-cli/issues/1565