arduino/arduino-cli #1613: `Segmentation fault` when compiling for ESP8266

4ntoine, 2022-01-03:
## Bug Report

That's unclear whether it's a toolchain issue or arduino-cli.

### Current behavior

```
./arduino-cli compile --fqbn esp8266:esp8
266:nodemcuv2 --verbose ./sk1/
Using board 'nodemcuv2' from platform in folder: /root/app/data/packages/esp8266/hardware/esp8266/3.0.2
Using core 'esp8266' from platform in folder: /root/app/data/packages/esp8266/hardware/esp8266/3.0.2
Detecting libraries used...
/root/app/data/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRR
Error while detecting libraries included by /tmp/arduino-sketch-BDF2558F8AD63D50A456B689AE6148DE/sketch/sk1.ino.cpp
Generating function prototypes...
/root/app/data/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STR-

Error during build: signal: segmentation fault
```

### Expected behavior

Sketch is compiled

### Environment

- CLI version (output of `arduino-cli version`):
```
# ./arduino-cli version
arduino-cli  Version: 0.20.2 Commit: 13783819 Date: 2021-12-09T13:32:05Z
```
- OS and platform:
Alpine linux 3.15.0 virt aarch64

### Additional context

```
# cat ./sk1/sk1.ino 

void setup() {
}

void loop() {
}
# cat /etc/os-release
NAME="Alpine Linux"
ID=alpine
VERSION_ID=3.15.0
PRETTY_NAME="Alpine Linux v3.15"
HOME_URL="https://alpinelinux.org/"
BUG_REPORT_URL="https://bugs.alpinelinux.org/"
```

`gcompat` package is installed with `apk add gcompat`.

```
# ./arduino-cli core list
ID              Installed Latest Name   
esp8266:esp8266 3.0.2     3.0.2  esp8266
```

Here is how you can create a vm with qemu to reproduce:
```
# create a disk
qemu-img create -f raw asvm-arm64-hd.img 2G
# prepare the UFI and some vars images
dd if=/dev/zero of=efi.img bs=1048576 count=64
dd if=/usr/share/qemu-efi-aarch64/QEMU_EFI.fd of=efi.img conv=notrunc
# install the system to the drive
qemu-system-aarch64 -nographic -machine virt -m 512M -cpu cortex-a53 -smp 4 -drive file=asvm-arm64-3.15-hd.img,if=none,id=drive0,cache=writeback,format=raw -device virtio-blk,drive=drive0,bootindex=0 -drive file=alpine-virt-3.15.0-aarch64.iso,if=none,id=drive1,cache=writeback,format=raw -device virtio-blk,drive=drive1,bootindex=1 -drive file=efi-3.15.img,format=raw,if=pflash -device vmxnet3,netdev=net0 -netdev user,id=net0,hostfwd=tcp::2222-:22
#(reboot)
# configure
qemu-system-aarch64 -nographic -machine virt -m 512M -cpu cortex-a53 -smp 4 -drive file=asvm-arm64-3.15-hd.img,if=none,id=drive0,cache=writeback,format=raw -device virtio-blk,drive=drive0,bootindex=0 -drive file=efi-3.15.img,format=raw,if=pflash -device vmxnet3,netdev=net0 -netdev user,id=net0,hostfwd=tcp::2222-:22,hostfwd=tcp::50051-:50051 -monitor tcp::5555,telnet,server,nowait
	(configure the VM)
	vi /etc/apk/repositories (uncomment ../community)
	apk add gcompat
```

then `scp arduin-cli` to the vm and:
```
./arduino-cli core install esp8266:esp8266
```

4ntoine, 2022-01-03:
PS. Works as expected on x86_64, but crashes on aarch64

4ntoine, 2022-01-03:
Seems to be toolchain issue:
```
# /root/arduino_studio/data/packages/esp826
6/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++ -D__
ets__ -DICACHE_FLASH -U__STR
Segmentation fault
```

4ntoine, 2022-01-03:
However it seems it's expected to be working on Platformio ([link](https://github.com/platformio/platformio-core/issues/1263)). Repack or another toolchain build needed for arduino?

silvanocerza, 2022-01-04:
I have not tested it but doesn't seem like an Arduino CLI issue to me, that would mean we're installing the wrong binary but am pretty confident that the code that handles is rock solid.

Just to be sure could you try installing `arduino:mbed_portenta` and compile for `arduino:mbed_portenta:envie_m4` or `arduino:mbed_portenta:envie_m7`? 

If compilation doesn't fail it means `arduino-cli` is installing and running the correct tools, so your issue stems from the esp8266 core, else it's on us.

4ntoine, 2022-01-04:
Thank for looking into it, @silvanocerza. Yeah, it also seems like not ardino-cli issue to me. I suspected alpine which uses not glibc internally to be the reason. I will try it and report later, thanks.

4ntoine, 2022-01-04:
@silvanocerza that's really strange: i was unable to install `arduino:mbed_portenta`:
```
asvm-aarch64-a53:~/arduino_studio/bin# ./arduino-cli core install arduino:mbed_portenta &> log.txt
asvm-aarch64-a53:~/arduino_studio/bin# cat ./log.txt | more
Tool arduino:openocd@0.11.0-arduino2 already installed
Downloading packages...
arduino:arm-none-eabi-gcc@7-2017q4 already downloaded
arduino:dfu-util@0.10.0-arduino1 already downloaded
arduino:mbed_portenta@2.6.1 already downloaded
Installing arduino:arm-none-eabi-gcc@7-2017q4...
Error during install: Cannot install tool arduino:arm-none-eabi-gcc@7-2017q4: extracting archive: Create file /root/arduino_studio/data/tmp/package-631039740/gcc-arm-none-eabi-7-2018-q2-update/arm-na
```

4ntoine, 2022-01-04:
I've tried on Debian arm64 and it worked as expected. The issue remains on alpine linux but it's not arduino-cli issue indeed. Closing it.

closed 2022-01-04 by 4ntoine