arduino/arduino-cli #1617: More information about linking and segment sizes?

Lorandil, 2022-01-09:
Hi there,

I'm having trouble with my code size exploding.
After days of trying and testing, I realized that the linker seems to add a data block twice (I can provide an example, if required).
My point is, that I searched in a totally wrong place - I thought that the compiler tried some code inlining because my code size
grew by > 400 bytes after inserting a single 'pgm_read_byte()'.

This would have been much easier to detect, if the segment sizes would be listed in the output window after compilation.

Is there a possiblity to share this information? Or is there only a text segment?

per1234, 2022-01-09:
Thanks for taking the time to request this @Lorandil.

I see we have another report about this at https://github.com/arduino/arduino-cli/issues/480 (and even a related proposal at https://github.com/arduino/arduino-cli/pull/1211).

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-01-09 by per1234

Lorandil, 2022-01-09:
Hi per1234,

 

looking at my last proposals, you could think that I'm not using the search function... ;)

Truth is, I was using it, but obviously not with the correct terms - sorry for that!

 

Btw: Do you have any idea where I could ask for help regarding my problem with binary structures which appear to be linked in double?

 

Cheers,
Lorandil
 

Gesendet: Sonntag, 09. Januar 2022 um 08:58 Uhr

per1234, 2022-01-09:
>I was using it, but obviously not with the correct terms - sorry for that!

No worries.

>Do you have any idea where I could ask for help regarding my problem with binary structures which appear to be linked in double?

I think you are on the right track asking on the Arduino Forum. There are a lot of very knowledgeable helpers over there:
https://forum.arduino.cc/t/problem-with-linker-including-same-binary-structure-twice/945196