arduino/arduino-cli #1624: lib search force update-index

cavonlee, 2022-01-14:
## Bug Report

### Current behavior

./arduino-cli lib search --names --format jsonmini

Running lib search will always force to update index, and update index  is not outputing jsonmini as command options

### Expected behavior

Donot force to update index every time and outputing json exactly

### Environment

- CLI version (output of `arduino-cli version`): Version: 0.20.2 Commit: 13783819 Date: 2021-12-07T16:41:50Z
- OS and platform: macOS Monterey 12.1

### Additional context

core search works fine without forcing update index, outputing json exactly

per1234, 2022-01-14:
Related: https://github.com/arduino/arduino-cli/pull/1043, https://github.com/arduino/arduino-cli/pull/1586

cavonlee, 2022-01-14:
@per1234 OK， if update index is meant to do before every search, is fine, just a little slow. but the problem is that, the update index output format jsonmini is not working. and search outputs so many datas, it would be helpful to use jsonmini.

per1234, 2022-01-14:
Hi @cavonlee.  Thanks so much for your report. My comment was only intended to add some cross-references to related information.

I see that an index caching system was put in place for `core search` here: https://github.com/arduino/arduino-cli/pull/1237 but that was not done for `lib search`.

I can also reproduce your observation that the update index output is printed in text format with `--format jsonmini`. The same issue happens with `--format yaml`, but not with `--format json`. I believe the problem is here:
https://github.com/arduino/arduino-cli/blob/edc63f8381d53ba03515a5004dd03b2676eeb14d/cli/output/rpc_progress.go#L36

cavonlee, 2022-01-14:
@per1234 OK, thanks too. If so I wanna ask why update index every time search? it slows down the search process. And if so, the update index is becoming usless. I suggest break them up, and let user decide when to update. Or maybe break into two level api, the high level one do both, makes it easier for beginners, if it is what concidering.

per1234, 2022-01-24:
>the update index output format jsonmini is not working

Since the bug of not all output being valid JSON when `--format jsonmini` is used is completely separate from the one of an index update being done on every `lib search` command, I have moved it to a dedicated issue here: https://github.com/arduino/arduino-cli/issues/1635.

That will make it easier for us to track the two.

Thanks for your report @cavonlee!

closed 2023-02-13 by MatteoPologruto