arduino/arduino-cli #1627: Embed settings in the internal CoreInstance

silvanocerza, 2022-01-17:
Currently configs used by the Arduino CLI are stored in a global struct, both when running in `daemon` or command line mode.

This behaviour has some issues since it prevents gRPC consumers to use different configs for different clients connected to it, this means that if a client wants to change settings they change for all clients.

The solution is to embed the settings in the `CoreInstance` and access them only through it. This will enhance a lot how settings are handled and also make it easier to write unit tests for certain code parts.

silvanocerza, 2022-03-18:
Part of this task is already been done on branch `scerza/embedding-settings`.

umbynos, 2023-02-02:
[image]

Unfortunately this branch is quite old now, we are going for a similar approach in the future maybe.

closed 2023-02-02 by umbynos