arduino/arduino-cli #1652: Error detecting boards

jobalas, 2022-02-03:
## Bug Report

### Current behavior

The command `arduino-cli board list`

Returns the following error:

`Error detecting boards: Error starting board discoveries: [discovery builtin:mdns-discovery process not started: fork/exec /root/.arduino15/packages/builtin/tools/mdns-discovery/1.0.5/mdns-discovery: no such file or directory]
`

### Expected behavior

Get a list of connected boards

### Environment

arduino-cli  Version: 0.20.2 Commit: 13783819 Date: 2021-12-09T13:32:05Z

- OS and platform:

OS: Armbian buster (21.02.3) armv7l
running on Orange Pi Zero

using Arduino Nano board
### Additional context

Installed arduino-cli by using this script:
curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh

jobalas, 2022-02-03:
mdns-discovery file exists
`orangepizero:~:# ls /root/.arduino15/packages/builtin/tools/mdns-discovery/1.0.5
LICENSE.txt  mdns-discovery
`

kubriel, 2022-02-23:
i believe this is a problem i solved with linking:
ln -s /lib/ld-linux-armhf.so.3 /lib/ld-linux.so.3
hopefuly it helps somebody at some point.

umbynos, 2023-02-03:
It seems a problem similar to https://github.com/arduino/arduino-cli/issues/1208.
Linking the library should solve the problem.

cmaglie, 2023-03-17:
@umbynos you may check if the mdns-discovery could be built without CGO enabled.

umbynos, 2023-03-22:
solved in https://github.com/arduino/mdns-discovery/pull/47, released in https://github.com/arduino/mdns-discovery/releases/tag/v1.0.9

closed 2023-03-22 by umbynos