arduino/arduino-cli #1653: `START_SYNC` errors after command sequence indicated by specification

per1234, 2022-02-04:
## Describe the bug

From the Pluggable Discovery Specification:

https://arduino.github.io/arduino-cli/dev/pluggable-discovery-specification/#start-command

> The `START` command initializes and starts the discovery internal subroutines. This command must be called before `LIST` or `START_SYNC`.

To me, this indicates that in order to execute a `START_SYNC`, I should run the following sequence of commands:

1. `HELLO`
1. `START`
1. `START_SYNC`

:bug: The sequence of commands indicated by the specification results in an error from the `START_SYNC` command.

## To Reproduce

```
$ git clone https://github.com/arduino/pluggable-discovery-protocol-handler

$ cd pluggable-discovery-protocol-handler/dummy-discovery/

$ go build

$ ./dummy-discovery
HELLO 1 "arduino-cli"
{
  "eventType": "hello",
  "message": "OK",     
  "protocolVersion": 1 
}
START
{
  "eventType": "start",
  "message": "OK"      
}
START_SYNC
{
  "eventType": "start_sync",
  "message": "Discovery already STARTed, cannot START_SYNC",
  "error": true
}
```

## Expected behavior

Specification to be corrected if the behavior of [the `github.com/arduino/pluggable-discovery-protocol-handler/v2` module](https://github.com/arduino/pluggable-discovery-protocol-handler) is what is intended, which would entail one of the following:

- Remove the statement that `START` is required before `START_SYNC`
- Specify that `STOP` must be called before `START_SYNC`

**-OR-**

The `github.com/arduino/pluggable-discovery-protocol-handler/v2` module to support the command sequence indicated in the specification.

## Desktop

- OS: Windows 10, Ubuntu 20.04
- github.com/arduino/pluggable-discovery-protocol-handler/v2@https://github.com/arduino/pluggable-discovery-protocol-handler/commit/a0d87f340f91cef6454ea0601f55d1119a5a8bb7

per1234, 2022-02-04:
Thanks for taking a look Cristian. I transferred the issue to the Arduino CLI repo since it is now clear that the defect must be corrected in this repository.

per1234, 2022-02-04:
We will also need to resolve the other instances of the error that propagated from the specification to the documentation of other projects:

- https://github.com/arduino/pluggable-discovery-protocol-handler/tree/main/dummy-discovery#start-command
- https://github.com/arduino/serial-discovery#start-command

I will propose that this be done by replacing the duplicate content in those repositories with a link to the Pluggable Discovery Specification document in order to avoid the need to maintain multiple copies.

PaulStoffregen, 2022-02-04:
Having recently implemented Pluggable Discovery, how to properly begin START_SYNC mode was one of the places I applied some intuition and guesswork.  Please forgive me if this is too much for this relatively simple issue (words in START command don't agree the State Machine)....

The main question in my mind is regarding a possible (though unlikely) race between the protocol commands and hardware changes.  Based only on the spec, a client like arduino-cli would need to use the LIST command first to learn of all the currently connected devices.  Then it would issue STOP to return to the idle state, and then START_SYNC to begin tracking changes.  But what happens if hardware changes after the last LIST command, but before START_SYNC begins?

As a proactive and perhaps paranoid (maybe I worry about this stuff too much) measure, Teensy's implementation of START_SYNC transmits a set of eventType add messages for all currently known devices when entering START_SYNC mode.  I believe elsewhere the spec says redundant "add" events are allowed and are to be interpreted as updating existing info.  But maybe even this is not good enough?  What happens if hardware is removed during that time between the last LIST and entering START_SYNC mode?  Would arduino-cli forever believe that hardware is still present, because it was connected at the time of the last LIST, and a "remove" eventType message is never transmitted because it's already gone by the time START_SYNC is supposed to transmit real-time changes.

PaulStoffregen, 2022-02-04:
Also, if you're updating the spec, please consider the wording "A well behaved pluggable discovery tool must reflect the following state machine".  Perhaps the word "must" really applies to the client which communicates with the discovery tool.  The client is the entity which "must" obey this state machine (eg, only transmit START_SYNC while in the "Idling" state).

PR #1749 (2022-06-06): Fixed error in pluggable discovery spec
**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [ ] Tests for the changes have been added (for bug fixes / features)
- [x] Docs have been added / updated (for bug fixes / features)
- [ ] `UPGRADING.md` has been updated with a migration guide (for breaking changes)

**What k [... cut]
commit 78d013bf (2022-06-06): [skip-changelog] Fixed error in pluggable discovery spec (#1749)

Fix #1653
closed 2022-06-06 by cmaglie