arduino/arduino-cli #1658: Sybolic Links in <sketches>/libraries shows up in using list with destination of link directory

KurtE, 2022-02-13:


ubidefeo, 2023-02-03:
@KurtE 
What happens is that the symlink is expanded, and when collecting paths and resources, the real final path is displayed.
This is not really by design but seems to me like a great way to know what resources are really being used.

That said, when you want to compile against a development library, 
you can simply invoke the CLI passing in a `--library` option followed by a path.
This is how I do it

```
arduino-cli compile -b arduino:samd:nano_33_iot --library /data/GITHUB/ArduinoLibs/FTDebouncer
```

this will ignore `FTDebouncer` in my **libraries** folder and prefer to use the one I pointed it too.
The scoring system for Library choice gives maximum ranking to the one specified using `--library` 

You also have a `--libraries` option folder which can be an extra **libraries** folder to point to, and all the libraries in it will supersede the ones in the Sketchbook's **libraries** folder

cmaglie, 2023-03-17:
@KurtE the symlink resolution is part of the "canonicalization" of the file name https://github.com/arduino/go-paths-helper/blob/269e91aa596347d22e73b253e54483842e51d530/paths.go#L530-L543
(see also https://github.com/golang/go/issues/17084#issuecomment-246645354)

We need to do this because sometimes a unique file name for a specific resource is needed (for example the language server requires it), so I'm closing this as "expect by design".

closed 2023-03-17 by cmaglie