arduino/arduino-cli #1660: Provide a way to identify board options via pluggable discovery

cmaglie, 2022-02-14:
As per @PaulStoffregen [comment](https://github.com/arduino/tooling-rfcs/pull/2#issuecomment-1029452622):

> How would you feel about extending Board Identification to allow JSON properties to match the menu options, so arduino-cli could know more of the FQBN?
> 
> VENDOR:ARCHITECTURE:BOARD_ID[:MENU_ID=OPTION_ID[,MENU2_ID=OPTION_ID ...]]
> 
> As I understand Board Identification, today arduino-cli can at best report the "VENDOR:ARCHITECTURE:BOARD_ID" portion of FQBN, because the spec only defines board.txt entries for matching the main part of FQBN.
> 
> If the Board Identification part of spec were extended slightly to specify boards.txt entries such as
> 
> {boardname}.menu.{menuname}.{menuoption}.upload_port.{identifier}=Value
> {boardname}.menu.{menuname}.{menuoption}.upload_port.#.{identifier}=Value
> 
> then a discovery tools capable of detecting which menu options where used could report properties to match those lines. Future arduino-cli could use this property matching to give a more complete FQBN for the detected hardware. Then in an even farther future, the IDE could utilize the more specific FQBN to initialize those menu options when a user clicks the port+board from the toolbar's drop-down list.

I think it could be implemented exactly as described above.

PaulStoffregen, 2022-06-09:
@cmaglie - Now with IDE 2.0-rc7 supporting pluggable monitor, I'm revisiting my package for Teensy to better integrate with IDE 2.0.  Any chance some of this can get put into the [Pluggable Discovery Specification](https://arduino.github.io/arduino-cli/0.23/pluggable-discovery-specification/) with notes about future implementation?  My hope is to implement it soon in my discovery tool, and sometime in the future CLI & IDE 2.0 will make use of the extra info.

cmaglie, 2022-06-15:
Hi @PaulStoffregen 

I've actually implemented it here: https://github.com/arduino/arduino-cli/pull/1674
It's rebased on the current master, so it's up-to-date, but I've just barely tested, so weird things may happen...
If you want to try it here are the test builds: https://github.com/arduino/arduino-cli/actions/runs/2501986948

There are still some task to be done:
* add the documentation
* add more tests
* review the code

Considering my current priorities, this won't happen before July...

ubidefeo, 2022-08-05:
@cmaglie @PaulStoffregen any updates on this one?

PaulStoffregen, 2022-08-12:
@ubidefeo - I implemented this for Teensy's "USB Type" menu in version 1.57.0 of teensy-discovery.  This is the package index to install 1.57.0, if you want to give it a try.

https://www.pjrc.com/teensy/td_156/package_teensy_index.json

Sadly, I haven't had time to add USB descriptors and code in teensy-discovery for Teensy's other 3 menus.  Been crazy busy here dealing with supply chain problems.  But teensy-discovery 1.57.0 definitely does support a "usbtype" property and package 1.57.0 has corresponding lines in boards.txt needed to build the FQBN with that menu setting, as in the screenshot above.

PaulStoffregen, 2022-08-12:
@per1234 - I see you tagged "status: waiting for information".  I don't understand what more information could be needed?  From @cmaglie's message on June 15, looks like we're waiting on 3 specific tasks before #1674 can be merged.

closed 2022-10-18 by cmaglie