arduino/arduino-cli #1683: arduino-cli: not found. Sopine A64 on OpenWrt OS

NamlaZz, 2022-03-03:
I am unable to install any version of arduino-cli beyond 0.5.0-showports.

This is what I get when installing through curl the latest version of arduino-cli. Note that /root/bin/ is already appended to my $PATH.
`root@s200:~# curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh`
`Installing in /root/bin`
`ARCH=ARM64`
`OS=Linux`
`Using curl as download tool`
`Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.21.1_Linux_ARM64.tar.gz`
`sh: /root/bin/arduino-cli: not found`
`Failed to install arduino-cli`

However when installing version 0.5.0-showports or older it works:
`root@s200:~# curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh -s 0.5.0-showports`
`Installing in /root/bin`
`ARCH=ARM64`
`OS=Linux`
`Using curl as download tool`
`Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.5.0-showports_Linux_ARM64.tar.gz`
`arduino-cli Version: 0.5.0-showports Commit: 3d7717e installed successfully in /root/bin`

Version 0.5.0-showports does not contain all the functionalities that I need for the project, most importantly the serial monitor. Attached below are system information for my OS.

**Linux version 5.10.96 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r18729-8c1d5129ee) 11.2.0, GNU ld (GNU Binutils) 2.37)**

per1234, 2022-03-03:
Hi @NamlaZz. Thanks for your report.

What do you get if you run this command?:

```text
ls /root/bin/arduino-cli
```

NamlaZz, 2022-03-03:
Running this command after failure to install is shown as below:

`sh: /root/bin/arduino-cli: not found`
`Failed to install arduino-cli`
`root@s200:~# ls /root/bin/arduino-cli`
`/root/bin/arduino-cli`

per1234, 2022-03-03:
I'm pretty sure the installation script is failing at this command:

https://github.com/arduino/arduino-cli/blob/12d488fc8a805a4653ed173a3a7e174da4963f91/install.sh#L205

What do you get if you run this command?:

```text
/root/bin/arduino-cli version
```

NamlaZz, 2022-03-03:
this is what I get when I run the command:
`root@s200:~# /root/bin/arduino-cli version`
`-ash: /root/bin/arduino-cli: not found`

umbynos, 2022-03-04:
What shell are you using? Do you have a `/tmp` folder on your system?

facchinm, 2022-03-04:
I think the problem is the missing `glibc` in your OS (since it's built on `musl`)
AFAIK the cli shouldn't have any CGO dependency on Linux so, if compiled with `CGO_ENABLED=0` it should also run on your setup.

Would you mind testing the attached binary? It's the cli compiled with `GOOS=linux GOARCH=arm64 CGO_ENABLED=0 go build`
[arduino-cli-cgo-disabled.zip](https://github.com/arduino/arduino-cli/files/8187119/arduino-cli-cgo-disabled.zip)

facchinm, 2022-03-04:
For reference, this is the output on `file` on both binaries
```
file ./arduino-cli-official                     
./arduino-cli: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, Go BuildID=v0P18lFJNQ4WeE4mt508/fNlonX00LtFjhF7tzJe9/SC8S6mPmPVKw-PyFgZrm/GTK9qrY1RiFtclTFe8Ot, not stripped
file /tmp/arduino-cli-cgo-disabled                  
/tmp/arduino-cli: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), statically linked, Go BuildID=Z9HDT5q9nZy8QNtP6vDG/RVYDkO_0ZmzWrUrdlDvG/RkARVf0-69-GYy4yltRQ/Jr4KPJrso4moT-71H9xI, not stripped
```

So the difference is indeed `statically linked` vs `dynamically linked`

NamlaZz, 2022-03-04:
> What shell are you using? Do you have a `/tmp` folder on your system?

it's Almquist shell, I have `/tmp` folder

NamlaZz, 2022-03-04:
> I think the problem is the missing `glibc` in your OS (since it's built on `musl`) AFAIK the cli shouldn't have any CGO dependency on Linux so, if compiled with `CGO_ENABLED=0` it should also run on [...]

I SCPed the file into `/root/bin` and indeed it works, I tried running the below command to check the version.
`root@s200:~# arduino-cli version`
`arduino-cli  Version: 0.0.0-git Commit:  Date: `

`A new release of Arduino CLI is available: 0.0.0-git → 0.21.1`
`https://arduino.github.io/arduino-cli/latest/installation/#latest-packages`

although not indicated it appears that I have a working version of 0.20 or 0.21 of arduino-cli. What can I do to bypass this problem for future releases?

facchinm, 2022-03-07:
My suggestion for the @arduino/team_tooling is to compile with `CGO_ENABLED=0` for all the platform where it's possible (probably every platform except MacOS)