arduino/arduino-cli #1684: Allow boards.local.txt to stay local

hpssjellis, 2022-03-03:
Problem: When using boards.local.txt you need to move the file to the board software location beside the boards.txt file. This location is not accessible from many schools, making several advanced programs not able to be run.

When that code is finished compiling we need to then delete the boards.local.txt file for things to go back to working normally.

Example try running 

https://github.com/edgeimpulse/firmware-arduino-portenta-h7

Feature Request:

Have the IDE search the local sketch folder for a boards.local.txt file before compiling.

per1234, 2022-03-03:
Thanks for your suggestion @hpssjellis.

Although I think this would be appropriate for other use cases (most likely as another path to global `#define`s a la https://github.com/arduino/arduino-builder/issues/15), for your specific use case I think that a custom boards platform would be far superior.

With the custom boards platform approach, the user can simply select the appropriate board from the **Tools > Board** menu in the Arduino IDE, as well as making more granular configurations via [custom board option](https://arduino.github.io/arduino-cli/latest/platform-specification/#custom-board-options) menus if needed. The Arduino boards platform system allows you to reference resources ([core libraries](https://arduino.github.io/arduino-cli/latest/platform-specification/#cores), [core variants](https://arduino.github.io/arduino-cli/latest/platform-specification/#core-variants), [tool recipes](https://arduino.github.io/arduino-cli/latest/platform-specification/#tools)) from other installed platforms

Information about that here:

https://arduino.github.io/arduino-cli/latest/platform-specification/#referencing-another-core-variant-or-tool

Examples of minimal boards platforms using referencing:

- https://github.com/arduino/ArduinoCore-Z80-MBC
- https://github.com/damellis/attiny
- https://github.com/carlosefr/atmega

(these happen to be for AVR, but the system can be used with any architecture)

If you have any questions about creating such a platform, just ask over on the Arduino forum and I'll provide assistance there.

Hyratel, 2022-05-21:
can I put a `boards.local.txt` directly in the sketch folder,? and if not, Why? This needs a good reason for it to not be present or intuitive. it's right there in the name - 'local' so intuitively, I should be able to add a boards.local.txt with a custom, single-project board def without adding it to Sketchbook/hardware

matthijskooijman, 2022-05-21:
The original reason I added `boards.local.txt` and `platform.local.txt` a long time ago, was to have a git checkout of a core, and be able to customize it with a local file without having to make changes to a git-tracked file. "local" in this context was intended to mean "not in git".

I would oppose allowing a `boards.local.txt` or `platform.local.txt` in the sketch folder, since modifying these files allows modifying the compiler commands that are run during build. If you would read these files from a sketch directory, this would create a security problem when you compile an untrusted sketch, which could then lead to arbitrary command execution.

See for example also https://github.com/arduino/Arduino/issues/421#issuecomment-326992418 which is an older discussion of this potential security issue.

Hyratel, 2022-05-21:
if we take the Arduino IDE's design at face value ie, the basic user should never have to step outside the arduino ecosystem environment/user-space folders, having to put `*.local.txt` directly into the library folders in `/Appdata/` rather misses the bus.

my example case here is I'm using a chip (non-AVR, exacts are contextually irrelevant) with several manufacturer variant suffixes, and the libraries I'm working with define a `Generic` of the Lowest Denominator/Most Limited version of the chip, and I wish to work with a higher-spec version of the chip (ver: 16k Flash -> ver: 32k Flash). I have found where in the boards.txt this is defined but I don't know of any way to add a menu entry *from within the Arduino Userspace Directories* without completely remaking the entire `/hardware/` stack in `/Sketches/Hardware/`

if such a method does exist, then it's not documented anywhere I can easily discover with clear keywords for searchability (and 'hunt through every page on the wiki' is a tall order)

matthijskooijman, 2022-05-21:
> if we take the Arduino IDE's design at face value ie, the basic user should never have to step outside the arduino ecosystem environment/user-space folders, having to put *.local.txt directly into the library folders in /Appdata/ rather misses the bus.

Agreed, but that mostly means that these basic users are not the intended audience of `*.local.txt` files (instead, like I said, the audience is power users that use Arduino cores from git...).

> if such a method does exist, then it's not documented anywhere I can easily discover with clear keywords for searchability (and 'hunt through every page on the wiki' is a tall order)

I'm not sure I understand your case exactly, but I *think* that creating a boards-only core could be helpful for your usecase. This means a core that contains just a `boards.txt` (and a small info-only `platform.txt`), that contains no recipes and not `cores/arduino` directory but inherits these from some existing other core. This is also the option that @per1234 describes above, with some example links.

umbynos, 2023-02-03:
Closing, because documented alternative exist.

closed 2023-02-03 by umbynos

pdcook, 2023-02-04:
> Closing, because documented alternative exist.

There's a documented alternative for specifying compiler flags from within the sketch's directory? I apologize for necro-ing this thread, but if that's the case would you be able to point me to said feature/documentation? I would greatly appreciate it!

per1234, 2023-02-04:
@pdcook "alternative" refers to the use case specified by **hpssjellis**. This is about the ability for the user to easily switch back and forth between a modified platform configuration and the stock official platform configuration.

As [I mentioned previously](https://github.com/arduino/arduino-cli/issues/1684#issuecomment-1058434618), for that use case, there is a documented alternative to the proposal. This is to create a minimal platform that contains only the necessary modification and otherwise references all resources from the parent platform is already provided by the Arduino platform framework. This alternative is more user friendly due to allowing the switch to be made from the user interface instead of copying configuration files into the sketches, and avoids Arduino sketches becoming an attack vector as [described by **matthijskooijman**](https://github.com/arduino/arduino-cli/issues/1684#issuecomment-1133770034).

As for specifying compiler flags as part of the sketch, there is a proposal here: https://github.com/arduino/tooling-rfcs/issues/9. Links to some previous alternative proposals are listed under https://github.com/arduino/arduino-cli/issues/846.