arduino/arduino-cli #1687: Failed to install arduino-cli on macOS Monterey

istvanzk, 2022-03-07:
## Bug Report
Installation fails when run with the install script:
`curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=~/bin sh`

It has been working up to version 0.18. None of the newer versions work and yield the same error.

### Current behavior

`curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=~/bin sh`

Output is:
`
Installing in /Users/izk/bin
ARCH=64bit
OS=macOS
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.21.1_macOS_64bit.tar.gz
Failed to install arduino-cli
`

### Expected behavior

Installation and arduino-cli binary in the ~/bin folder.

### Environment

- CLI version (output of `arduino-cli version`): Not possible to run, see below.
- OS and platform: macOS Monterey 12.2.1, on MacBook Pro (2018)

### Additional context
I have also tried to install the pre-built binary directly (arduino-cli_0.21.1_macOS_64bit.tar.gz)
When running the command `arduino-cli version`
the output is `Killed: 9`

ubidefeo, 2022-03-22:
@istvanzk 
on M1 I have tested the install script and it won't work work because it looks for `arm64`.
The downloaded release file works fine.
We'll try and test on an Intel machine and keep you posted

istvanzk, 2022-03-23:
Correct. I'm using MBP with Intel processor:
Model Name:	MacBook Pro
  Model Identifier:	MacBookPro15,3
  Processor Name:	6-Core Intel Core i7
  Processor Speed:	2,6 GHz
  Number of Processors:	1
  Total Number of Cores:	6
  L2 Cache (per Core):	256 KB
  L3 Cache:	9 MB
  Hyper-Threading Technology:	Enabled
  Memory:	32 GB

per1234, 2022-04-01:
>It has been working up to version 0.18. None of the newer versions work and yield the same error.

So that I can be sure I understand, please tell me which of the two versions is the oldest one with this problem:

- 0.18.0
- 0.18.1

>When running the command `arduino-cli version`
the output is `Killed: 9`

The install script also runs an `arduino-cli version` command, so this is likely the cause of the installation failure as well.

["9" is the signal number of `SIGKILL`](https://man7.org/linux/man-pages/man7/signal.7.html#:~:text=Signal%20numbering%20for%20standard%20signals)

istvanzk, 2022-04-01:
Thank you. The installation worked for version 0.18.1. I have not tried out other version in-between 0.18.1 and current one.

istvanzk, 2022-04-30:
Any updates on this? Any hints where/what should I check? Or is simply an incompatibility due to latest OS?

istvanzk, 2022-05-09:
For completeness: exactly the same error happens with the latest release arduino-cli_0.22.0_macOS_64bit.
I.e. the download and unpacking works fine, while the `arduino-cli version` fails with `Killed: 9`

istvanzk, 2022-05-09:
Also for completness: I have tried to remove the _com.apple.quarantine_ attribute with
`xattr -d com.apple.quarantine arduino-cli`
This did still not solve the error.

istvanzk, 2022-05-09:
And a more positive update: I have removed the old binary (actually the latest v0.22) with `rm arduino-cli` and then re-run the installation with `curl ...`. Magically, this did not fail and the output of the `arduino-cli version` was `arduino-cli  Version: 0.22.0 Commit: 65f662a7 Date: 2022-05-06T11:09:35Z`
It seems like Mac OS quarantined the old binary (at some point) and later even after removing the _com.apple.quarantine_ attribute from the new binary, somehow the binary was still treated as being the _old_. The `rm` command did magic :)

closed 2022-05-09 by istvanzk

per1234, 2022-05-09:
Thanks for taking the time to update the issue with your solution @istvanzk. I'm very glad that Arduino CLI is working for you now. I'll definitely keep this information in mind in case there are other reports of that "`Killed: 9`" error.