arduino/arduino-cli #1689: Problem programming Arduino Micro

davo08, 2022-03-09:
There is a one second delay after finding the new port (after the touch_1200 reset).  The comment says it is needed for OS X.  On Linux, I find that 95% of the time the boot loader has already timed out by the time the programming is ready to start.  This could be an issue with the boot loader, of course, but rebuilding arduino-cli with that delay removed resolves my problem.

### Current Behavior

```
$ arduino-cli upload -p /dev/ttyACM0 -b arduino:avr:micro
Connecting to programmer:
```
Then it hangs.

### Expected behavior

With the delay removed:

```
$ arduino-cli upload -p /dev/ttyACM0 -b arduino:avr:micro
Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44
```
and the sketch is uploaded and started correctly.

### Environment

- CLI version: arduino-cli  Version: 0.21.1 Commit: 9fcbb392 Date: 2022-02-24T15:41:45Z

- OS and platform: Ubuntu 18.04

WestfW, 2022-03-10:
Is there another delay somewhere (ie in avrdude?)  Normally the bootloader delay will be substantially higher than 1s, since that delay starts AFTER the LED blinking.  (well, it does in optiboot.  I guess Caterina could be different.)

davo08, 2022-03-10:
In case the bootloader had an unusually short delay, I just tried reflashing the bootloader using an Arduino UNO as the programmer using my local build from here with version:

arduino-cli  Version: git-snapshot Commit: 8b53b850 Date: 2022-03-09T23:17:15Z

Retrying to upload a sketch with the delay still in place, I see the same issue.  Running with the -v option added, I see:

```
Performing 1200-bps touch reset on serial port /dev/ttyACM0
Waiting for upload port...
Upload port found on /dev/ttyACM0
".../.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude" "-C/.../.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf" -v -V -patmega32u4 -cavr109 "-P/dev/ttyACM0" -b57600 -D "-Uflash:w:/tmp/arduino-sketch-0D24644961BE97692F1B57E28CD74AAD/BlinkMe.ino.hex:i"
```

There does seem to be a delay in avrdude, and preexisting sketch is back running again before it gets very far.

per1234, 2022-03-10:
I can't reproduce the problem. Uploading is successful to my Micro with the same command. I burned the bootloader to it to make sure we are using the same one.

OS: Ubuntu 20.04
Arduino CLI:  nightly-20220310 Commit: 8b53b85 Date: 2022-03-10T01:33:54Z

davo08, 2022-03-10:
I tried on a machine with Ubuntu 20.04 and another with 18.04 and had no problem on either of those.  I guess there's something about the original machine I tried this on that made it more sensitive to the extra delay.

jepperaskdk, 2022-10-13:
Same issue here:
```
$ uname -r
5.4.0-128-generic

$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.6 LTS
Release:        18.04
Codename:       bionic

$ arduino --version
Picked up JAVA_TOOL_OPTIONS:
Loading configuration...
Initializing packages...
Preparing boards...
Arduino: 1.8.19

$ arduino-cli version
arduino-cli  Version: 0.27.1 Commit: a900cfb2 Date: 2022-09-06T16:44:27Z
```

duke8804, 2024-05-05:
Is there any hope of this being resolved. The red text makes me think it never works.

[image]

per1234, 2024-05-05:
> Same issue

@jepperaskdk by "same issue", do you mean that you find that uploads are failing due to the reported fault?:

> 95% of the time the boot loader has already timed out by the time the programming is ready to start

per1234, 2024-05-05:
@duke8804 the red text is not the fault being reported here. The fault being reported here is:

> 95% of the time the boot loader has already timed out by the time the programming is ready to start

If uploads are completing successfully for you, then you are not affected by this very rare fault.

As for the red text, Arduino CLI and Arduino IDE simply display whatever text is printed by the upload tool (in this case [**AVRDUDE**](https://avrdudes.github.io/avrdude/)). Tools may print text on two different "streams": "standard output" (AKA "stdout") and "standard error" (AKA "stderr"). Since it is traditionally used for warning and error messages, Arduino IDE colors all text received from the "standard error" stream red. The AVRDUDE developers made the [decision](https://www.nongnu.org/avrdude/environment.html#:~:text=all%20messages%20of%20any%20kind%20go%20to%20stderr) to print even the tool's purely informational output to the "standard error" stream, which is why this text is colored red.

Since Arduino CLI and Arduino IDE are general purpose development tools designed to be used with any of the ever increasing thousands of different boards in the Arduino ecosystem, each of which might use a different upload tool, nothing can or should be done in the Arduino codebases about this so reports of it are off topic here. As for reporting it to AVRDUDE, they made a conscious decision to design the tool this way based on their opinion of how the "standard output" and "standard error" streams should be utilized so I don't think there is any chance they would be willing to change the tool's behavior. All this is to say that you just have to get used to the red text. Arduino IDE will display a notification at the end of the upload. If the upload was successful, the notification will say "**Done uploading.**". If the upload failed, the notification will say "**Upload error: ...**".