arduino/arduino-cli #1691: `arduino-cli compile` does not respect sketch.json

nalzok, 2022-03-19:
Here is a reproducible example

```
$ arduino-cli sketch new test
Sketch created in: /tmp/test
$ cd test/
$ arduino-cli board attach -p /dev/cu.usbmodem14101
Board found: Adafruit Feather ESP32-S2 TFT...
Selected fqbn: esp32:esp32:adafruit_feather_esp32s2_tft
$ cat sketch.json
{
  "cpu": {
    "fqbn": "esp32:esp32:adafruit_feather_esp32s2_tft",
    "name": "Adafruit Feather ESP32-S2 TFT",
    "port": "serial:///dev/cu.usbmodem14101"
  }
}
 $ arduino-cli compile --build-path build -v test.ino
Using board 'adafruit_feather_esp32s2_tft' from platform in folder: /Users/qys/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2
Using core 'esp32' from platform in folder: /Users/qys/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2
<irrelevant output truncated>
Sketch uses 217374 bytes (16%) of program storage space. Maximum is 1310720 bytes.
Global variables use 17748 bytes (5%) of dynamic memory, leaving 309932 bytes for local variables. Maximum is 327680 bytes.
$ arduino-cli compile --build-path build -u -v test.ino
Error during FQBN detection: Missing FQBN (Fully Qualified Board Name)
```

In the last line, I deliberately wrote `arduino-cli compile --build-path build -u -v test.ino` without `-b esp32:esp32:adafruit_feather_esp32s2_tft -p /dev/tty.usbmodem14101` to test if `arduino-cli compile` can find the missing information in the `sketch.json` file. As you can see, it throws an error.

https://github.com/arduino/arduino-cli/blob/47c634bd910f3f6efc9f3fd2e530048d6e90c5dd/cli/compile/compile.go#L165-L179

Perhaps "board autodetection" should check for `sketch.json` first?

matthijskooijman, 2022-03-19:
I've also been running into this, for at least a few months now. Commandlines that used to work stopped working at some point, but I've never taken the time to bisect this or figure out under what conditions it doesn't work (it seems when doing `compile --upload`, I think a plain `--compile` does work). But I can confirm this problem exists (and I can also reliably reproduce it).

per1234, 2022-10-26:
Thanks so much for reporting this bug @nalzok. I am able to reproduce the bug using versions from 0.21.0 (I didn't check older versions) to 7f2e750d5285111da0b28ccac255bb37022a23ab, but not with any newer revision.

I bisected the fix to https://github.com/arduino/arduino-cli/pull/1717

Since it was the one mentioned here, I only tested with `arduino-cli compile --upload`. If anyone discovers other commands that do not use the data from `sketch.json`, please submit a bug report.

closed 2022-10-26 by per1234