arduino/arduino-cli #1703: Can't upload to my generic UNO

chaquit0, 2022-04-02:
## Bug Report

### Current behavior
[neewbs]
Board: Generic --- ELEGOO UNO R3 (**ATMEGA328P U**) 

This command:   
`sudo arduino-cli upload -p /dev/ttyACM0 -b
arduino:avr:uno BlinkMe`
Gives this result: 
`Error during Upload: Platform 'arduino:avr' not found: platform arduino:avr
is not installed
Try running arduino-cli core install arduino:avr`

-Output of `arduino-cli core list` :
```
ID          Installed Latest Name
arduino:avr 1.8.5     1.8.5  Arduino AVR Boards
```

### Expected behavior
Upload the code to the ATMEGA328P

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli  Version: 0.21.1 Commit: 9fcbb392 Date: 2022-02-24T15:41:45Z
`
- OS and platform: 
```
- OS: Artix Linux x86_64 
- Kernel: 5.26.16-artix-1
- CPU:  Intel i7-2630QM (8) @2.900G
- GPU: Intel 2nd Gen Core
- 
```

### Additional context
I followed all the steps from the official website as well as the Official YT channel, the core that I'm using is already installed !!

per1234, 2022-04-02:
> ```text
> sudo arduino-cli upload -p /dev/ttyACM0 -b arduino:avr:uno BlinkMe
> ```

Hi @chaquit0. The problem will be solved by removing the `sudo` from the command.

closed 2022-04-02 by per1234

chaquit0, 2022-04-02:
Hi, @per1234 thanks for answering.
If I don´t `sudo`  this is the output:
 `avrdude: ser_open(): can't open device "/dev/ttyACM0": Permission denied
Error during Upload: Failed uploading: uploading error: exit status 1`
that's why I use it, also I `sudo su` so I can run the command as you tell me, and this is the output: 
` Error during Upload: Platform 'arduino:avr' not found: platform arduino:avr is not installed
Try running  arduino-cli core install arduino:avr`
The core is already installed, I tried other core version (1.6.5) and the same error happened, Don't know what's wrong,  :( . Thanks for your time again.

per1234, 2022-04-03:
This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/c/using-arduino/avrdude-stk500-bootloader-issues/81