arduino/arduino-cli #1714: Installation script's github downloader blows up due to shell line length issue on macos 12.3

obra, 2022-04-15:
## Bug Report

### Current behavior

When running the current install script:
`curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh |sh`

It blows up because it doesn't find an arduino download for `arm64` and then falls back to scraping the GitHub releases page for a download. The GitHub releases page for 0.21.1 has more than 4k of content on it, leading to a `bash` line length failure.

### Expected behavior

I'd expect the install to still fail due to the arm64 / 64bit issue.

### Environment

- CLI version (output of `arduino-cli version`): 0.21.1
- OS and platform: macos12.3 (M1)

### Additional context

There's a pretty straightforward change that would work for the arduino-cli implementation, although it might not be fully portable to other places arduino uses this script. You don't need to parse the releases page, since the release urls are easily predictable.

```
   # Support specifying nightly build versions (e.g., "nightly-latest") via the script argument.
   case "$TAG" in
   nightly*)
     DOWNLOAD_URL="https://downloads.arduino.cc/${PROJECT_NAME}/nightly/${APPLICATION_DIST}"
     ;;
   *)
     DOWNLOAD_URL="https://downloads.arduino.cc/${PROJECT_NAME}/${APPLICATION_DIST}"
    
 +GITHUB_TAG_DOWNLOAD_URL="https://github.com/arduino/${PROJECT_NAME}/releases/download/${TAG}/${PROJECT_NAME}/${APPLICATION_DIST}"
     ;;
   esac
 
   INSTALLATION_TMP_FILE="/tmp/$APPLICATION_DIST"
   echo "Downloading $DOWNLOAD_URL"
   httpStatusCode=$(getFile "$DOWNLOAD_URL" "$INSTALLATION_TMP_FILE")
   if [ "$httpStatusCode" -ne 200 ]; then
+    echo "Did not find a release for your system published on downloads.arduino.cc: $OS $ARCH"
+   echo "Trying to find a release from the GitHub releases page."
+   httpStatusCode=$(getFile "$GITHUB_TAG_DOWNLOAD_URL" "$INSTALLATION_TMP_FILE")
```  

This is what the failure looks like.

```
ARCH=arm64
OS=macOS
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.21.1_macOS_arm64.tar.gz
Did not find a release for your system: macOS arm64
Trying to find a release using the GitHub API.
LATEST_RELEASE_URL=https://api.github.com/repos/arduino/arduino-cli/releases/tags/0.21.1
Getting https://api.github.com/repos/arduino/arduino-cli/releases/tags/0.21.1
sh: line 454: list table in case of unknown board (#1673)

## Fixed

9fcbb392 Fix error being printed in index parsing during first start (#1672)
e6285efd Fix a couple memory leaks caused by files not being closed (#1670)

## Changed

d684dec1 Board list now returns partial results in case of errors (#1668)
625ff52b Discoveries are now closed and unregistered after failure (#1667)",
  "reactions": {
    "url": "https://api.github.com/repos/arduino/arduino-cli/releases/60384143/reactions",
    "total_count": 5,
    "+1": 5,
    "-1": 0,
    "laugh": 0,
    "hooray": 0,
    "confused": 0,
    "heart": 0,
    "rocket": 0,
    "eyes": 0
  }
}: File name too long
```

per1234, 2022-04-15:
Thanks for your report @obra .

We maintain the base content of this reusable asset at https://github.com/arduino/tooling-project-assets/tree/main/other/installation-script

I see we have another report about this in that repository at https://github.com/arduino/tooling-project-assets/issues/207.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

I have already transferred your valuable suggestion over to that thread to make sure we will not lose track of it.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-04-15 by per1234

obra, 2022-04-15:
This is a different issue.  This is a bug in github release parsing that’s unrelated to the M1 issue. But happy to have it tracked in the right repo.

per1234, 2022-05-02:
Thanks for pointing that out @obra.

I made fixes for both https://github.com/arduino/tooling-project-assets/issues/207 and the separate bug you reported here and those fixes have now been pulled into this repository by https://github.com/arduino/arduino-cli/pull/1721