arduino/arduino-cli #1715: Arduino board not discoverable

jjshoe, 2022-04-20:
## Bug Report

### Current behavior

`arduino-cli board list`

gives me:

```
Error detecting boards: Error starting board discoveries: [starting discovery builtin:serial-discovery: calling START: timeout waiting for message from builtin:serial-discovery]
No boards found.
```

### Expected behavior

I expect to see my board listed

### Environment

arduino-cli  Version: 0.21.1 Commit: 9fcbb392 Date: 2022-02-24T15:41:45Z

ubuntu 20.04 
kernel 5.4.0-65-generic

### Additional context

lsusb `Bus 001 Device 006: ID 2341:0062 Arduino SA Arduino UNO Mini`

syslog:
```
Apr 20 18:20:22 joel-desktop kernel: [350033.669698] usb 1-1: new full-speed USB device number 7 using xhci_hcd
Apr 20 18:20:22 joel-desktop kernel: [350033.947797] usb 1-1: New USB device found, idVendor=2341, idProduct=0062, bcdDevice= 0.02
Apr 20 18:20:22 joel-desktop kernel: [350033.947799] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=220
Apr 20 18:20:22 joel-desktop kernel: [350033.947801] usb 1-1: Product: Arduino UNO Mini
Apr 20 18:20:22 joel-desktop kernel: [350033.947801] usb 1-1: Manufacturer: Arduino (www.arduino.cc)
Apr 20 18:20:22 joel-desktop kernel: [350033.947802] usb 1-1: SerialNumber: 851383136333518161D1
Apr 20 18:20:22 joel-desktop kernel: [350033.963433] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
Apr 20 18:20:22 joel-desktop mtp-probe: checking bus 1, device 7: "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-1"
Apr 20 18:20:22 joel-desktop mtp-probe: bus: 1, device: 7 was not an MTP device
Apr 20 18:20:22 joel-desktop snapd[268447]: hotplug.go:199: hotplug device add event ignored, enable experimental.hotplug
Apr 20 18:20:22 joel-desktop mtp-probe: checking bus 1, device 7: "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-1"
Apr 20 18:20:22 joel-desktop mtp-probe: bus: 1, device: 7 was not an MTP device
```

/dev/ttyACM0 exists: `crw-rw---- 1 root dialout 166, 0 Apr 20 18:20 /dev/ttyACM0`

and `groups` shows me in the dialout group.

jjshoe, 2022-04-20:
Occasionaly it works, only randomly, I've tried different ports, cables, etc. 

I rebooted once, and it worked for a bit. Syslog ALWAYS sees it when I unplug/plug in, I always get the tty device. No idea why arduino-cli doesn't.

cmaglie, 2022-06-03:
Hi @jjshoe,

is still happening?
may you update to the latest CLI, and re-run with the `-v` flag to see more logging?

umbynos, 2023-02-07:
Hi @jjshoe, closing for lack of feedback, If you need assistance feel free to reopen this

closed 2023-02-07 by umbynos