arduino/arduino-cli #1718: arduino-cli Command Completion results in errors

zfields, 2022-04-21:
## Bug Report

### Current behavior

I can tab complete the command `arduino-cli` itself, but attempting to tab complete parameters results in error

```
# ard <tab>
# arduino-cli <tab><tab>
```

```
# arduino-cli bash: _get_comp_words_by_ref: command not found
```

### Expected behavior

I would expect to see a list of parameters, but not an error message.

### Environment

- CLI version (output of `arduino-cli version`): 0.21.1
- OS and platform: Linux

### Additional context

I've created a docker container to facilitate a repro.

```Dockerfile
# Copyright 2022 Blues Inc.  All rights reserved.
# Use of this source code is governed by licenses granted by the
# copyright holder including that found in the LICENSE file.

# Build development environment
# docker build --tag arduino-buildpack .

# Launch development environment
# docker run --device /dev/bus/usb/ --interactive --rm --tty --user blues --volume "$(pwd)":/host-volume/ arduino-buildpack bash

# Define global arguments
ARG DEBIAN_FRONTEND="noninteractive"
ARG UID=1000
ARG USER="blues"

# POSIX compatible (Linux/Unix) base image
FROM debian:stable-slim

# Import global arguments
ARG DEBIAN_FRONTEND
ARG UID
ARG USER

# Define local arguments
# ARG ARDUINO_CLI_CHECKSUM="068f0a69ffecd5688e0b4bc02609720af57ec1aa60e5ea6a2879af3ed1dc0d9b"
ARG ARDUINO_CLI_VERSION=0.21.1
ARG BINDIR=/usr/local/bin
ARG ECHO_BC_FILE='$bcfile'

# Define environment variables
ENV ARDUINO_UPDATER_ENABLE_NOTIFICATION=false

# Create Non-Root User
RUN ["dash", "-c", "\
    addgroup \
     --gid ${UID} \
     \"${USER}\" \
 && adduser \
     --disabled-password \
     --gecos \"\" \
     --ingroup \"${USER}\" \
     --uid ${UID} \
     \"${USER}\" \
 && usermod \
     --append \
     --groups \"dialout,plugdev\" \
     \"${USER}\" \
"]

# Establish development environment
RUN ["dash", "-c", "\
    apt-get update --quiet \
 && apt-get install --assume-yes --no-install-recommends --quiet \
     bash \
     ca-certificates \
     curl \
 && apt-get clean \
 && apt-get purge \
 && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/* \
"]

# Download/Install Arduino CLI
RUN ["dash", "-c", "\
    curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=${BINDIR} sh -s ${ARDUINO_CLI_VERSION} \
# && arduino-cli completion bash > /usr/share/bash-completion/completions/arduino-cli.sh \
 && mkdir -p /etc/bash_completion.d/ \
 && arduino-cli completion bash > /etc/bash_completion.d/arduino-cli.sh \
 && echo >> /etc/bash.bashrc \
# && echo \"for bcfile in /usr/share/bash-completion/completions/* ; do\" >> /etc/bash.bashrc \
 && echo \"for bcfile in /etc/bash_completion.d/* ; do\" >> /etc/bash.bashrc \
 && echo \"    [ -f \\\"${ECHO_BC_FILE}\\\" ] && . \\\"${ECHO_BC_FILE}\\\"\" >> /etc/bash.bashrc \
 && echo \"done\" >> /etc/bash.bashrc \
"]

# Set Execution Environment
WORKDIR /host-volume

CMD ["bash"]
```

zfields, 2022-04-22:
@per1234 I'm not sure how this is an `imperfection`. The command completion functionality does NOT appear to work at all.

per1234, 2022-04-22:
@zfields the label name is intentionally chosen to be applicable to any severity of defect on the spectrum from "missing comma in the readme" to "Skynet has declared war on the human race".

We have [a separate set of "**criticality**" labels](https://github.com/arduino/arduino-cli/labels?q=criticality) which are used to indicate severity. "**criticality**" labels are assigned by a manager during the planning process, while "**type**" labels are assigned by a maintainer during triage.

zfields, 2022-04-22:
I understand now. Thank you for explaining. :pray:

umbynos, 2022-06-06:
Hi @zfields,
apparently, you need to install [`bash-completion`](https://packages.debian.org/buster/bash-completion) on Debian slim. And also enable bash completion in `/etc/bash.bashrc` by uncommenting:
``` bash
# enable bash completion in interactive shells
#if ! shopt -oq posix; then
#  if [ -f /usr/share/bash-completion/bash_completion ]; then
#    . /usr/share/bash-completion/bash_completion
#  elif [ -f /etc/bash_completion ]; then
#    . /etc/bash_completion
#  fi
#fi
```

I've applied that suggestion to your docker example:
``` diff
--- <unnamed>
+++ <unnamed>
@@ -54,6 +54,7 @@
      bash \
      ca-certificates \
      curl \
+     bash-completion \
  && apt-get clean \
  && apt-get purge \
  && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/* \
@@ -62,14 +63,15 @@
 # Download/Install Arduino CLI
 RUN ["dash", "-c", "\
     curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=${BINDIR} sh -s ${ARDUINO_CLI_VERSION} \
-# && arduino-cli completion bash > /usr/share/bash-completion/completions/arduino-cli.sh \
  && mkdir -p /etc/bash_completion.d/ \
  && arduino-cli completion bash > /etc/bash_completion.d/arduino-cli.sh \
  && echo >> /etc/bash.bashrc \
-# && echo \"for bcfile in /usr/share/bash-completion/completions/* ; do\" >> /etc/bash.bashrc \
  && echo \"for bcfile in /etc/bash_completion.d/* ; do\" >> /etc/bash.bashrc \
  && echo \"    [ -f \\\"${ECHO_BC_FILE}\\\" ] && . \\\"${ECHO_BC_FILE}\\\"\" >> /etc/bash.bashrc \
  && echo \"done\" >> /etc/bash.bashrc \
+ && echo \"if [ -f /etc/bash_completion ]; then\" >> /etc/bash.bashrc \
+ && echo \" . /etc/bash_completion\" >> /etc/bash.bashrc \
+ && echo \"fi\" >> /etc/bash.bashrc \
 "]
 
 # Set Execution Environment
```

zfields, 2022-06-26:
@umbynos, sorry it took me so long to test your suggestion. It worked perfectly!

Thank you so much!

closed 2022-06-26 by zfields