arduino/arduino-cli #1722: error while loading shared libraries

Assowavesss, 2022-05-02:
boy@nasty:~/Documents/arduino_project/SB/SafeBox$ arduino-cli compile --fqbn arduino:samd:mkrwifi1010 .
Sketch uses 26212 bytes (9%) of program storage space. Maximum is 262144 bytes.
Global variables use 3832 bytes (11%) of dynamic memory, leaving 28936 bytes for local variables. Maximum is 32768 bytes.

boy@nasty:~/Documents/arduino_project/SB/SafeBox$ arduino-cli upload -p /dev/ttyACM0 --fqbn arduino:samd:mkrwifi1010 .
/home/boy/snap/arduino-cli/19/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3/bossac: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
Error during Upload: Failed uploading: uploading error: exit status 127

Assowavesss, 2022-05-02:
It's okay to compile but when uploading ....., plus, I use the library WiFiNINA and I instal it as well

per1234, 2022-05-02:
Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there.

I see you already made a post:
https://forum.arduino.cc/t/arduino-cli-esp32-not-working-throwing-error-error-while-loading-shared-libraries-libstdc-so-6-cannot-open-shared-object-file-no-such-file/987200

closed 2022-05-02 by per1234