arduino/arduino-cli #1752: Add support for defining a default build profile

per1234, 2022-06-08:
### Describe the request

Implement [the "**Using a default profile**" component](https://github.com/arduino/tooling-rfcs/blob/main/RFCs/0003-build-profiles.md#using-a-default-profile) of [the "**Sketch build profiles and reproducible builds**" RFC](https://github.com/arduino/tooling-rfcs/blob/main/RFCs/0003-build-profiles.md) has now been implemented (https://github.com/arduino/arduino-cli/pull/1713).
🙂

### Describe the current behavior

The majority of the "**Sketch build profiles and reproducible builds**" RFC has now been implemented (https://github.com/arduino/arduino-cli/pull/1713).

One missing part is the support for [the `default_profile` key](https://github.com/arduino/tooling-rfcs/blob/main/RFCs/0003-build-profiles.md#using-a-default-profile) of the [sketch project file](https://arduino.github.io/arduino-cli/dev/sketch-project-file/).

🙁 There is currently no way to define a default profile, to be used when the `arduino-cli` command operating on a sketch with build profiles does not contain a `--profile` flag.

### Arduino CLI version

0.23.0

### Operating system

All

### Operating system version

N/A

### Additional context

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

cmaglie, 2023-06-14:
Here are some use cases and the expected outcome:

## Sketch with the following project file:
```
profiles:
  uno:
    fqbn: arduino:avr:uno
    platforms:
      - platform: arduino:avr (1.8.6)
```

#### Case 1:
```
$ arduino-cli compile
```
Output: error `Missing FQBN (Fully Qualified Board Name)`

#### Case 2:
```
$ arduino-cli compile --profile uno
```
Output: compile using profile `uno`

#### Case 3:
```
$ arduino-cli compile --profile uno -b arduino:avr:leonardo
```
Output: compile using profile `uno` but overrides FQBN with `arduino:avr:leonardo`

#### Case 4:
```
$ arduino-cli compile -b arduino:avr:uno
```
Output: compile for `arduino:avr:uno` without using the profiles

## Sketch with the following project file:
```
profiles:
  uno:
    fqbn: arduino:avr:uno
    platforms:
      - platform: arduino:avr (1.8.6)

  leonardo:
    fqbn: arduino:avr:leonardo
    platforms:
      - platform: arduino:avr (1.8.6)

default_profile: leonardo
```

#### Case 5:
```
$ arduino-cli compile
```
Output: compiles using the default profile `leonardo`

#### Case 6:
```
$ arduino-cli compile --profile uno
```
Output: compiles using profile `uno`

#### Case 7:
```
$ arduino-cli compile --profile uno -b arduino:avr:leonardo
```
Output: compile using profile `uno` but overrides FQBN with `arduino:avr:leonardo`

#### Case 8:
```
$ arduino-cli compile -b arduino:avr:uno
```
Output: compile using the default profile `leonardo` but overrides FQBN with `arduino:avr:uno`

alexxus, 2023-08-16:
Please implement it. It would be so helpful!

I'm forced to use the `--profile` flag, even if my `sketch.yaml` file has only one profile. It doesn't make sense!
arduino-cli could simply default to the single profile.

Something like `default_profile: leonardo`, like suggested by @cmaglie could be also a good alternative. In fact, it would be almost better than the "magical" defaulting to the single profile. In this case the user would tell, which profile has to be used as the default.

closed 2023-08-23 by MatteoPologruto