arduino/arduino-cli #1760: The order of the compiler errors are indeterministic

kittaakos, 2022-06-14:
### Describe the problem

If I have a sketch with compiler errors, as a user of the CLI, I would expect the order of the errors is deterministic. It does not.

### To reproduce

[Errors.zip](https://github.com/arduino/arduino-cli/files/8898273/Errors.zip)

 - Download and extract the attached `Errors.zip` sketch somewhere. 👆 
 - Compile with the CLI:
```
% ~/Downloads/arduino-cli compile -b arduino:mbed_nano:nanorp2040connect ~/Documents/Arduino/Errors
In file included from /Users/a.kitta/Documents/Arduino/Errors/CppTab.cpp:1:0:
/Users/a.kitta/Documents/Arduino/Errors/HTab.h:1:1: error: 'bazerror' does not name a type
 bazerror;
 ^~~~~~~~
/Users/a.kitta/Documents/Arduino/Errors/CppTab.cpp: In function 'void qux()':
/Users/a.kitta/Documents/Arduino/Errors/CppTab.cpp:3:3: error: 'quxerror' was not declared in this scope
   quxerror
   ^~~~~~~~
/Users/a.kitta/Documents/Arduino/Errors/CTab.c: In function 'pluto':
/Users/a.kitta/Documents/Arduino/Errors/CTab.c:2:3: error: 'plutoerror' undeclared (first use in this function); did you mean 'pluto'?
   plutoerror
   ^~~~~~~~~~
   pluto
/Users/a.kitta/Documents/Arduino/Errors/CTab.c:2:3: note: each undeclared identifier is reported only once for each function it appears in
/Users/a.kitta/Documents/Arduino/Errors/CTab.c:3:1: error: expected ';' before '}' token
 }
 ^
In file included from /Users/a.kitta/Documents/Arduino/Errors/Errors.ino:1:0:
/Users/a.kitta/Documents/Arduino/Errors/src/HFile.h:1:1: error: 'paperinoerror' does not name a type
 paperinoerror;
 ^~~~~~~~~~~~~
/Users/a.kitta/Documents/Arduino/Errors/Errors.ino: In function 'void setup()':
/Users/a.kitta/Documents/Arduino/Errors/Errors.ino:3:3: error: 'fooerror' was not declared in this scope
   fooerror
   ^~~~~~~~
/Users/a.kitta/Documents/Arduino/Errors/Errors.ino:3:3: note: suggested alternative: 'ferror'
   fooerror
   ^~~~~~~~
   ferror
/Users/a.kitta/Documents/Arduino/Errors/Errors.ino: In function 'void loop()':
/Users/a.kitta/Documents/Arduino/Errors/Errors.ino:6:3: error: 'barerror' was not declared in this scope
   barerror
   ^~~~~~~~
/Users/a.kitta/Documents/Arduino/Errors/Errors.ino:6:3: note: suggested alternative: 'strerror'
   barerror
   ^~~~~~~~
   strerror
/Users/a.kitta/Documents/Arduino/Errors/InoTab.ino: In function 'void pippo()':
/Users/a.kitta/Documents/Arduino/Errors/InoTab.ino:2:3: error: 'pippoerror' was not declared in this scope
   pippoerror
   ^~~~~~~~~~
/Users/a.kitta/Documents/Arduino/Errors/InoTab.ino:2:3: note: suggested alternative: 'perror'
   pippoerror
   ^~~~~~~~~~
   perror

Used platform     Version Path                                                                      
arduino:mbed_nano 3.1.1   /Users/a.kitta/Library/Arduino15/packages/arduino/hardware/mbed_nano/3.1.1

Error during build: exit status 1
```

 - Without doing anything, repeat the previous step:
```
% ~/Downloads/arduino-cli compile -b arduino:mbed_nano:nanorp2040connect ~/Documents/Arduino/Errors
/Users/a.kitta/Documents/Arduino/Errors/CTab.c: In function 'pluto':
/Users/a.kitta/Documents/Arduino/Errors/CTab.c:2:3: error: 'plutoerror' undeclared (first use in this function); did you mean 'pluto'?
   plutoerror
   ^~~~~~~~~~
   pluto
/Users/a.kitta/Documents/Arduino/Errors/CTab.c:2:3: note: each undeclared identifier is reported only once for each function it appears in
/Users/a.kitta/Documents/Arduino/Errors/CTab.c:3:1: error: expected ';' before '}' token
 }
 ^
In file included from /Users/a.kitta/Documents/Arduino/Errors/CppTab.cpp:1:0:
/Users/a.kitta/Documents/Arduino/Errors/HTab.h:1:1: error: 'bazerror' does not name a type
 bazerror;
 ^~~~~~~~
/Users/a.kitta/Documents/Arduino/Errors/CppTab.cpp: In function 'void qux()':
/Users/a.kitta/Documents/Arduino/Errors/CppTab.cpp:3:3: error: 'quxerror' was not declared in this scope
   quxerror
   ^~~~~~~~
In file included from /Users/a.kitta/Documents/Arduino/Errors/Errors.ino:1:0:
/Users/a.kitta/Documents/Arduino/Errors/src/HFile.h:1:1: error: 'paperinoerror' does not name a type
 paperinoerror;
 ^~~~~~~~~~~~~
/Users/a.kitta/Documents/Arduino/Errors/Errors.ino: In function 'void setup()':
/Users/a.kitta/Documents/Arduino/Errors/Errors.ino:3:3: error: 'fooerror' was not declared in this scope
   fooerror
   ^~~~~~~~
/Users/a.kitta/Documents/Arduino/Errors/Errors.ino:3:3: note: suggested alternative: 'ferror'
   fooerror
   ^~~~~~~~
   ferror
/Users/a.kitta/Documents/Arduino/Errors/Errors.ino: In function 'void loop()':
/Users/a.kitta/Documents/Arduino/Errors/Errors.ino:6:3: error: 'barerror' was not declared in this scope
   barerror
   ^~~~~~~~
/Users/a.kitta/Documents/Arduino/Errors/Errors.ino:6:3: note: suggested alternative: 'strerror'
   barerror
   ^~~~~~~~
   strerror
/Users/a.kitta/Documents/Arduino/Errors/InoTab.ino: In function 'void pippo()':
/Users/a.kitta/Documents/Arduino/Errors/InoTab.ino:2:3: error: 'pippoerror' was not declared in this scope
   pippoerror
   ^~~~~~~~~~
/Users/a.kitta/Documents/Arduino/Errors/InoTab.ino:2:3: note: suggested alternative: 'perror'
   pippoerror
   ^~~~~~~~~~
   perror

Used platform     Version Path                                                                      
arduino:mbed_nano 3.1.1   /Users/a.kitta/Library/Arduino15/packages/arduino/hardware/mbed_nano/3.1.1

Error during build: exit status 1
```

First error with the first compile command:
```
/Users/a.kitta/Documents/Arduino/Errors/HTab.h:1:1: error: 'bazerror' does not name a type
```

First error with the second compile attempt:
```
/Users/a.kitta/Documents/Arduino/Errors/CTab.c:2:3: error: 'plutoerror' undeclared (first use in this function); did you mean 'pluto'?
```

### Expected behavior

No matter how many times I compile the same broken sketch, the output on the stderr is the same.

### Arduino CLI version

Version: nightly-20220614 Commit: 76fab32 Date: 2022-06-14T08:23:46Z

### Operating system

macOS

### Operating system version

12.3.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

umbynos, 2023-02-07:
Hi @kittaakos, unfortunately this is not a bug. The compiler is run concurrently depending on how many CPUs you have. In this case, two parallel compilations fail at the same time (`CTab.c` and `CppTab.cpp`). We do not have control on the output of the compiler.

kittaakos, 2023-02-07:
Using the same attached sketch, I ran 20 consecutive verify with the same core for the same board. The order of the compiler errors was always the same. I did the compilation on the same machine. What is the difference?

<details><summary>Click to see the output of the 20 Verify with Arduino Nano RP 2040 Connect with Arduino IDE 1.8.19</summary>
<p>

```
Arduino: 1.8.19 (Mac OS X), Board: "Arduino Nano RP2040 Connect"

/Users/a.kitta/Desktop/Errors/CTab.c: In function 'pluto':
CTab.c:2:3: error: 'plutoerror' undeclared (first use in this function); did you mean 'pluto'?
   plutoerror
   ^~~~~~~~~~
   pluto
/Users/a.kitta/Desktop/Errors/CTab.c:2:3: note: each undeclared identifier is reported only once for each function it appears in
CTab.c:3:1: error: expected ';' before '}' token
 }
 ^
exit status 1
'plutoerror' undeclared (first use in this function); did you mean 'pluto'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

```
Arduino: 1.8.19 (Mac OS X), Board: "Arduino Nano RP2040 Connect"

/Users/a.kitta/Desktop/Errors/CTab.c: In function 'pluto':
CTab.c:2:3: error: 'plutoerror' undeclared (first use in this function); did you mean 'pluto'?
   plutoerror
   ^~~~~~~~~~
   pluto
/Users/a.kitta/Desktop/Errors/CTab.c:2:3: note: each undeclared identifier is reported only once for each function it appears in
CTab.c:3:1: error: expected ';' before '}' token
 }
 ^
exit status 1
'plutoerror' undeclared (first use in this function); did you mean 'pluto'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

```
Arduino: 1.8.19 (Mac OS X), Board: "Arduino Nano RP2040 Connect"

/Users/a.kitta/Desktop/Errors/CTab.c: In function 'pluto':
CTab.c:2:3: error: 'plutoerror' undeclared (first use in this function); did you mean 'pluto'?
   plutoerror
   ^~~~~~~~~~
   pluto
/Users/a.kitta/Desktop/Errors/CTab.c:2:3: note: each undeclared identifier is reported only once for each function it appears in
CTab.c:3:1: error: expected ';' before '}' token
 }
 ^
exit status 1
'plutoerror' undeclared (first use in this function); did you mean 'pluto'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

```
Arduino: 1.8.19 (Mac OS X), Board: "Arduino Nano RP2040 Connect"

/Users/a.kitta/Desktop/Errors/CTab.c: In function 'pluto':
CTab.c:2:3: error: 'plutoerror' undeclared (first use in this function); did you mean 'pluto'?
[... 314 lines cut ...]

```
Arduino: 1.8.19 (Mac OS X), Board: "Arduino Nano RP2040 Connect"

/Users/a.kitta/Desktop/Errors/CTab.c: In function 'pluto':
CTab.c:2:3: error: 'plutoerror' undeclared (first use in this function); did you mean 'pluto'?
   plutoerror
   ^~~~~~~~~~
   pluto
/Users/a.kitta/Desktop/Errors/CTab.c:2:3: note: each undeclared identifier is reported only once for each function it appears in
CTab.c:3:1: error: expected ';' before '}' token
 }
 ^
exit status 1
'plutoerror' undeclared (first use in this function); did you mean 'pluto'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

</p>
</details>

Thanks!

cmaglie, 2023-03-17:
Ensure that you have a syntax error in `CTab.c` **and** another syntax error in `CppTab.cpp`.

Since those two files should be compiled in parallel the order of the compile failure is not deterministic: it may fail `CTab.c` first (so the errors of `CTab.c` will be displayed first) or it may fail `CppTab.cpp` first (so the errors of `CppTab.cpp` are displayed first).

Anyway, it may be just difficult to reproduce...