arduino/arduino-cli #1761: The CLI might put the same error multiple times to the stderr

kittaakos, 2022-06-14:
### Describe the problem

When compiling a broken sketch, the CLI might dump the error location multiple times to the std err. It is most likely related to how the sketch files are preprocessed and the prototype is created. (Of course, I am just guessing here 😊)

### To reproduce

`sketch_jun14c.ino`:
```cpp
void setup(){}
void loop() {
  byte = 2;
}
```

```
Users/a.kitta/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino: In function 'void loop()':
/Users/a.kitta/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino:3:8: error: expected unqualified-id before '=' token
   byte = 2;
        ^
```

----- 

`sketch_jun14c.ino`:
```cpp
void setup(){}
void 1loop() {
  byte = 2;
}
```

```
/Users/a.kitta/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino:2:6: error: expected unqualified-id before numeric constant
 void 1loop() {
      ^~~~~
```

----

`sketch_jun14c.ino`:
```cpp
void setup(){}
xvoid loop() {
  byte = 2;
}
```

```
/Users/a.kitta/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino:2:1: error: 'xvoid' does not name a type; did you mean 'void'?
 xvoid loop() {
 ^~~~~
 void
/Users/a.kitta/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino:2:1: error: 'xvoid' does not name a type; did you mean 'void'?
 xvoid loop() {
 ^~~~~
 void
```

<details><summary>Click to see full console output</summary>
<p>

```
a.kitta@Akoss-MacBook-Pro build % ~/Downloads/arduino-cli version                                                                         
arduino-cli  Version: nightly-20220614 Commit: 76fab32 Date: 2022-06-14T08:23:46Z
a.kitta@Akoss-MacBook-Pro build % cat ~/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino                                                 
void setup(){}
void loop() {
  byte = 2;
}%                                                                                                                                                              a.kitta@Akoss-MacBook-Pro build % ~/Downloads/arduino-cli compile -b arduino:mbed_nano:nanorp2040connect ~/Documents/Arduino/sketch_jun14c
/Users/a.kitta/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino: In function 'void loop()':
/Users/a.kitta/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino:3:8: error: expected unqualified-id before '=' token
   byte = 2;
        ^

Used platform     Version Path                                                                      
arduino:mbed_nano 3.1.1   /Users/a.kitta/Library/Arduino15/packages/arduino/hardware/mbed_nano/3.1.1

Error during build: exit status 1
a.kitta@Akoss-MacBook-Pro build % cat ~/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino                                                 
void setup(){}
void 1loop() {
  byte = 2;
}%                                                                                                                                                              a.kitta@Akoss-MacBook-Pro build % ~/Downloads/arduino-cli compile -b arduino:mbed_nano:nanorp2040connect ~/Documents/Arduino/sketch_jun14c
/Users/a.kitta/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino:2:6: error: expected unqualified-id before numeric constant
 void 1loop() {
      ^~~~~

Used platform     Version Path                                                                      
arduino:mbed_nano 3.1.1   /Users/a.kitta/Library/Arduino15/packages/arduino/hardware/mbed_nano/3.1.1

Error during build: exit status 1
a.kitta@Akoss-MacBook-Pro build % cat ~/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino                                                 
void setup(){}
xvoid loop() {
  byte = 2;
}%                                                                                                                                                              a.kitta@Akoss-MacBook-Pro build % ~/Downloads/arduino-cli compile -b arduino:mbed_nano:nanorp2040connect ~/Documents/Arduino/sketch_jun14c
/Users/a.kitta/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino:2:1: error: 'xvoid' does not name a type; did you mean 'void'?
 xvoid loop() {
 ^~~~~
 void
/Users/a.kitta/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino:2:1: error: 'xvoid' does not name a type; did you mean 'void'?
 xvoid loop() {
 ^~~~~
 void

Used platform     Version Path                                                                      
arduino:mbed_nano 3.1.1   /Users/a.kitta/Library/Arduino15/packages/arduino/hardware/mbed_nano/3.1.1

Error during build: exit status 1
```

</p>
</details>

### Expected behavior

I would expect to see the following error once:
```
/Users/a.kitta/Documents/Arduino/sketch_jun14c/sketch_jun14c.ino:2:1: error: 'xvoid' does not name a type; did you mean 'void'?
 xvoid loop() {
 ^~~~~
 void
```

### Arduino CLI version

arduino-cli  Version: nightly-20220614 Commit: 76fab32 Date: 2022-06-14T08:23:46Z

### Operating system

macOS

### Operating system version

12.3.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-14:
As you suspected, the issue can be understood more clearly by looking at the C++ code that is generated from the `.ino` file by [the Arduino sketch preprocessor](https://arduino.github.io/arduino-cli/dev/sketch-build-process/#pre-processing):

```text
$ arduino-cli compile --fqbn arduino:avr:uno --preprocess C:/Users/per/Documents/Arduino/sketch_jun14c
#include <Arduino.h>
#line 1 "C:\\Users\\per\\Documents\\Arduino\\sketch_jun14c\\sketch_jun14c.ino"
#line 1 "C:\\Users\\per\\Documents\\Arduino\\sketch_jun14c\\sketch_jun14c.ino"
void setup();
#line 2 "C:\\Users\\per\\Documents\\Arduino\\sketch_jun14c\\sketch_jun14c.ino"
xvoid loop();
#line 1 "C:\\Users\\per\\Documents\\Arduino\\sketch_jun14c\\sketch_jun14c.ino"
void setup(){}
xvoid loop() {
  byte = 2;
}
```

[`#line` directives](https://gcc.gnu.org/onlinedocs/cpp/Line-Control.html) are used to cause the messages generated from compiling that C++ file to match the line numbering of the source `.ino` file. This means there are actually two lines of code which are identified as line two:

The generated [function prototype](https://cplusplus.com/doc/tutorial/functions/#declarations):

```cpp
xvoid loop();
```

and the first line of the function definition:

```cpp
xvoid loop() {
```

cmaglie, 2023-04-26:
May be resolved if we create a parser for the generated errors: https://github.com/arduino/arduino-cli/issues/1121

If we can parse the errors, we could easily add a de-duplicate.