arduino/arduino-cli #1762: As a gRPC consumer of the CLI, I need better error codes to distinguish between errors

kittaakos, 2022-06-14:
### Describe the request

To keep the startup performance OK, the [IDE2](https://github.com/arduino/arduino-ide) does not run a core and library index update before initializing the gRPC client. The Arduino core client initialization in the IDE2 happens the following way:
 - create the gRPC core client,
 - init the core client,
 - run the index update asynchronously.

However, this does not always work. The command line CLI auto-magically updates the core and lib indexes on the first run (https://github.com/arduino/arduino-cli/pull/1043). This is not available when consuming the CLI's capabilities via the gRPC APIs.

On the first start (`directories/data` is missing or empty), the gRPC init step will fail with the following errors:
```json
[
  { "code": 9, "message": "Loading index file: loading json index file /Users/a.kitta/Library/Arduino15/package_index.json: open /Users/a.kitta/Library/Arduino15/package_index.json: no such file or directory" },
  { "code": 9, "message": "Error loading hardware platform: discovery builtin:serial-discovery not found" },
  { "code": 9, "message": "Error loading hardware platform: discovery builtin:mdns-discovery not found" },
  { "code": 9, "message": "Loading index file: reading library_index.json: open /Users/a.kitta/Library/Arduino15/library_index.json: no such file or directory" }
]
```
When the IDE2 sees these errors, IDE2 knows that this is the first start, and the core and lib indexes must be updated before running the init phase of the gRCP core client. The above error is recoverable in IDE2.

----

There are "ignorable" errors. For example, if I use the `https://arduino.esp8266.com/stable/package_esp8266com_index.json` 3rd party URL, and install the `3.0.0` version (see the steps [here](https://github.com/arduino/arduino-ide/issues/1036#issuecomment-1152449530)), then there will be an [error](https://github.com/arduino/arduino-cli/blob/76fab32d9b758ecc38a29f7c75be805ea412f9ea/rpc/cc/arduino/cli/commands/v1/commands.proto#L216) status:
```json
{ "code": 9, "message": "Error loading hardware platform: loading platform release esp8266:esp8266@3.0.0: loading boards: skipping loading of boards esp8266:esp8266:espduino: malformed custom board options" }
```
This error does not cause any CLI error, so should be ignored.

### Describe the current behavior

All errors the IDE2 receives from the CLI have `code: 9`. There is no way to detect if the error is recoverable, non-recoverable, or ignorable.

It would be great if some details about the error would be available.

### Arduino CLI version

0.23.0

### Operating system

macOS

### Operating system version

12.3.1

### Additional context

Related: https://github.com/arduino/arduino-cli/pull/1251

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

kittaakos, 2022-12-16:
Besides the application-level error handling for the services, I would like to get a proper exit code and a terminating daemon process when something went wrong. Now, IDE2 must sniff into the stderr of the process, parse such messages and kill the daemon process because it's not terminating, although the daemon is nonfunctional:

 - To figure out the `directories.data` is unusable, IDE2 must parse this.
    ```
    invalid path creating config dir: /alma error: mkdir /alma: read-only file system time="2022-12-16T11:07:49+01:00" level=error msg="Error: invalid path creating config dir: /alma error: mkdir /alma: read-only file system"
    ```

 - To determine a `Failed to listen on TCP port. Address already in use.` error message:
    `failed to listen`, `address already in use`, `bind`,  `only one usage of each socket address`

 - For invalid address: `Failed to listen on TCP port. Unknown address.`:
    `is unknown name` or `tcp/` and `is an invalid port`

 - To provide the `Failed to listen on TCP port. Invalid port.` to the user:
    `is an invalid port` and `failed to serve`

Please help IDE2, terminate the daemon process on unrecoverable errors, such as inaccessible data dir or port issues, and provide an exit code. IDE2 wants to help users understand what went wrong, but it's hard because IDE2 is also just guessing the error. Thanks!

kittaakos, 2022-12-16:
> invalid path creating config dir: /alma error: mkdir /alma: read-only file system time="2022-12-16T11:07:49+01:00" level=error msg="Error: invalid path creating config dir: /alma error: mkdir /alma: read-only file system"

And it's not even creating any config dir, the `directories.data` is incorrect.

umbynos, 2023-01-23:
This requires some more brainstorming with @cmaglie

umbynos, 2023-02-06:
For now we will fix only this error. If there is a need to fix other errors, this is the moment to speak. We will follow the same pattern used already another error ("upload require programmer error"), we should use the same methodology.

kittaakos, 2023-02-06:
> this is the moment to speak.

> For now we will fix only this error.

IDE2 wants to avoid parsing error messages from the CLI as it's unreliable.

kittaakos, 2023-02-28:
I am going to reopen this issue as a generic go-to ticket until the CLI can provide error codes for at least the following issues:
 - the sketch is invalid
 - the sketch is missing
 - the sketch is not verified
 - no FQBN
 - invalid FQBN
 - the platform is not installed
 - the dependent platform is not installed when installing a 3rd party platform

kittaakos, 2023-03-01:
> For now we will fix only this error.

There are open questions on the initial changes at https://github.com/arduino/arduino-cli/pull/2076#issuecomment-1447892161.

umbynos, 2023-05-15:
Probably should not be necessary anymore after #2119

kittaakos, 2023-05-15:
> Probably should not be necessary anymore after #2119

It's an essential feature. Please see: https://github.com/arduino/arduino-cli/issues/1762#issuecomment-1447684498

> I am going to reopen this issue as a generic go-to ticket until the CLI can provide error codes for at least the following issues:
> 
> * the sketch is invalid
> * the sketch is missing
> * the sketch is not verified
> * no FQBN
> * invalid FQBN
> * the platform is not installed
> * the dependent platform is not installed when installing a 3rd party platform
> * verify has failed

kittaakos, 2023-11-13:
Please add a specific error code when the programmer is missing from `debug --info` (https://github.com/arduino/arduino-cli/pull/2394). Currently, clients must parse the stderr. This won't work when the CLI language is not English. Thanks!

kittaakos, 2023-11-16:
It's just partially done. See https://github.com/arduino/arduino-cli/issues/1762#issuecomment-1548255642.