arduino/arduino-cli #1765: Compile fails on symlink

econeale, 2022-06-14:
### Describe the problem

When executing `arduino-cli compile` within a project that is also a git repository, I receive this error: 

`/Users/neale/myProject/.git/hooks/pre-commit: no such file or directory`

This pre-commit hook is a symlink to a shell script that's also contained in the repository. This appears to happen regardless of where the symlink points / exists in the repository. There appears to be no way to work around the error other than completely removing the link. 

### To reproduce

Create a project with the structure: 

- myProject
  - myProject.ino
  - aLink -> /somewhere/else

Execute `arduino-cli compile` within the directory.  

### Expected behavior

Several solutions to this come to mind: 

1. Recursing hidden directories probably shouldn't happen by default
2. Ability to explicitly list files / directories to compile 
3. An option to follow / not follow symlinks 

### Arduino CLI version

0.23.0

### Operating system

macOS

### Operating system version

12.3.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

matthijskooijman, 2022-06-14:
Hm, I recall that .git (or maybe all hidden directories) were already filtered out from the list of files to compile/load, but maybe I remember wrong (or this was changed at some point, maybe).

In any case, I ran into issues with (broken) symlinks breaking compilation before, see #1438 for an attempt at fixing this (which is probably not ideal, but will be useful as a starting point and see some of the previous attempts, considerations and involved code).

cmaglie, 2023-04-26:
This requires two fix IMHO:
1. ignore `.git` folders in sketches
2. ignore broken symbolic links and continue the compilation anyway.

closed 2024-01-17 by cmaglie