arduino/arduino-cli #1774: Teensy Plugable Serial Monitor Working with new daily builds?

KurtE, 2022-06-18:


cmaglie, 2022-07-05:
You may try to manually run the teensy monitor and insert the commands to connect to the port, this way you can completely exclude the Arduino CLI. It's a bit tricky but doable, here is the procedure:

1) first of all you must run a program to listen to a TCP socket. Usually, on Linux/mac I use netcat, the command is `nc -l 12345` this will open a socket at port `12345`. Any random port is OK, AFAIK there should be also a netcat for Windows.
2) from another terminal run `teensy-monitor.exe`, the program should just wait for some inputs without any prompt. Enter:
  * `HELLO 1 "arduino-cli"`
  the monitor should answer with a confirmation message in JSON, after that enter:
  * `OPEN 127.0.0.1:12345 usb3/3-3`
  where `127.0.0.1:12345` is the IP:PORT where you are listening, and `usb3/3-3` is the address of the teensy port.
3) At this point the monitor should be proxying the data back and forth from the teensy to netcat.

If you see the correct output from netcat then the problem is in the Arduino CLI, otherwise, we must search for the bug in the teensy monitor.

KurtE, 2022-07-05:
@cmaglie  Thanks,

I installed ncat on my windows PC.  Running with two command prompts:

With the sketch:
```
elapsedMillis em = 0;
uint32_t alive_count = 0;
void setup() {
  pinMode(13, OUTPUT);
  while (!Serial)
    ;
  Serial.begin(115200);
  delay(1000);
  Serial.println("Start of Echo Test");
}

void loop() {
  if (Serial.available()) {
    digitalToggleFast(13);
    Serial.print("Received:(");
    int ch;
    while ((ch = Serial.read()) != -1) Serial.write(ch);
    Serial.println(")");
  }
  #if 1
  if (em > 5000) {
    // If I Uncomment the print below it echos???
    Serial.printf("Alive Count: %u\n", ++alive_count);
    em = 0;
  }
  #endif
}
```

So in the simple ncat terminal window:
```
C:\Program Files\Npcap>ncat -l 12345
Start of Echo Test
Alive Count: 1
Alive Count: 2
Alive Count: 3
abcdefg
Alive Count: 4
Received:(abcdefg
)
Alive Count: 5
...
```
In the other I ran:
```
C:\Users\kurte\AppData\Local\Arduino15\packages\teensy\tools\teensy-monitor\1.56.53>teensy-monitor.exe
HELLO 1 "arduino-cli"
{
"eventType": "hello",
"protocolVersion": 1,
"message": "OK"
}
OPEN 127.0.0.1:12345 usb:0/140000/0/1/2
{
"eventType": "open",
"message": "OK"
}
```

And in this case, like with IDE2, I receive the data I typed in, although delayed until the next Serial output.

But if I change the #if 1 to #if 0 
Such that it only prints out something if it received something, it never prints anything
```
C:\Program Files\Npcap>ncat -l 12345
Start of Echo Test
If at first
```
It did not echo. back.

I tried typing in several more lines of text, like 18 times the following line:
```
abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789
```
And still nothing echoed back.  Tried that in case it was something like waiting for a logical Serial flush operation 

Note: I built the sketch with USB Type: Serial

cmaglie, 2022-07-06:
Then, the problem is surely in the `teensy-monitor.exe`, since we have completely excluded arduino-cli from the test.
You have produced a very good test case, thanks for that!

closed 2022-07-06 by per1234

PaulStoffregen, 2022-07-23:
Quick followup, this was indeed a bug in teensy-monitor.exe.  It has been fixed in version 1.57-beta4.