arduino/arduino-cli #1775: "arduino-cli board list" returns "panic: runtime error: invalid memory address or nil pointer dereference"

Marcus-C137, 2022-06-19:
### Describe the problem

Raspberry pi 4 | Ubuntu 20.04 LTS
ARCH=ARMv7
OS=Linux

Installed via: 
curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh -s nightly-latest

Running: 
`arduino-cli board list`

Returns:
```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xb8e30]

goroutine 11 [running]:
os.(*Process).signal(0x0, {0xbb4280, 0x106018c})
	/usr/local/go/src/os/exec_unix.go:64 +0x38
os.(*Process).Signal(...)
	/usr/local/go/src/os/exec.go:138
os.(*Process).kill(...)
	/usr/local/go/src/os/exec_posix.go:68
os.(*Process).Kill(...)
	/usr/local/go/src/os/exec.go:123
github.com/arduino/arduino-cli/executils.(*Process).Kill(...)
	/home/build/executils/process.go:120
github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).killProcess(0x2010e80)
	/home/build/arduino/discovery/discovery.go:268 +0xdc
github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).Quit(0x2010e80)
	/home/build/arduino/discovery/discovery.go:381 +0x104
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).remove(0x1eb2130, {0x96f714, 0x16})
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:82 +0x94
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).RunAll.func1(0x2010e80)
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:131 +0x78
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize.func1(0x5c0aca0, 0x5b27d70, 0x465a8c0, 0x2010e80)
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:101 +0x50
created by github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:99 `+0x1ac`
```

### To reproduce

arduino-cli board list

### Expected behavior

Return board detected

### Arduino CLI version

arduino-cli  Version: nightly-20220618 Commit: 64bc4dc Date: 2022-06-18T01:37:19Z

### Operating system

Linux

### Operating system version

20.04LTS

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

andrewintw, 2022-06-27:
I go same issue on Banana Pi M2 Zero / Ubuntu 20.04  (Armbian)

I install arduino-cli on Armbian. The hardware is Banana Pi M2 Zero.

```
[16:20:36 ~] $ uname -a
Linux bananapim2zero 5.10.60-sunxi #21.08.1 SMP Wed Aug 25 18:19:32 UTC 2021 armv7l armv7l armv7l GNU/Linux

[16:31:59 ~] $ cat /etc/os-release 
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Armbian 21.08.1 Focal"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

I use a micro-USB hub to connect a arduino Nano. And it seems detected the board (USB driver is CH340)

```
[16:20:07 ~] $ lsusb 
Bus 001 Device 004: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter

[16:20:09 ~] $ ls -l /dev/ttyUSB0 
crw-rw---- 1 root dialout 188, 0 Jun 26 06:17 /dev/ttyUSB0
```

I didn't installed any library yet, just installed arduino-cli and then executed the `board list` command. Then I get an error message:

```
$ arduino-cli board list
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xb8e30]

goroutine 50 [running]:
os.(*Process).signal(0x0, {0xb8e5a8, 0x102018c})
	/usr/local/go/src/os/exec_unix.go:64 +0x38
os.(*Process).Signal(...)
	/usr/local/go/src/os/exec.go:138
os.(*Process).kill(...)
	/usr/local/go/src/os/exec_posix.go:68
os.(*Process).Kill(...)
	/usr/local/go/src/os/exec.go:123
github.com/arduino/arduino-cli/executils.(*Process).Kill(...)
	/home/build/executils/process.go:120
github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).killProcess(0x30371c0)
	/home/build/arduino/discovery/discovery.go:268 +0xdc
github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).Quit(0x30371c0)
	/home/build/arduino/discovery/discovery.go:381 +0x104
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).remove(0x2e6c040, {0x94b78e, 0x16})
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:82 +0x94
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).RunAll.func1(0x30371c0)
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:131 +0x78
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize.func1(0x7113a60, 0x70dcbc0, 0x71aa100, 0x30371c0)
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:101 +0x50
created by github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:99 +0x1ac
```

```
[16:19:38 ~] $ tree -a ~/.arduino15/
/home/andrew/.arduino15/
|-- inventory.yaml
|-- library_index.json
|-- library_index.json.sig
|-- package_index.json
|-- package_index.json.sig
|-- packages
|   `-- builtin
|       `-- tools
|           |-- ctags
|           |   `-- 5.8-arduino11
|           |       `-- ctags
|           |-- mdns-discovery
|           |   `-- 1.0.5
|           |       |-- LICENSE.txt
|           |       `-- mdns-discovery
|           |-- serial-discovery
|           |   `-- 1.3.2
|           |       |-- LICENSE.txt
|           |       `-- serial-discovery
|           `-- serial-monitor
|               `-- 0.9.1
|                   |-- LICENSE.txt
|                   `-- serial-monitor
|-- staging
`-- tmp

13 directories, 12 files
```

```
[16:20:40 ~] $ arduino-cli version
arduino-cli  Version: 0.24.0 Commit: c1b10f56 Date: 2022-06-22T10:09:11Z

[16:19:56 ~] $ arduino-cli lib list
No libraries installed.
```

andrewintw, 2022-06-27:
Here is the verbose output.

```
[21:09:34 ~] $ arduino-cli board list -v 
INFO[0000] Config file not found, using default values  
INFO[0000] arduino-cli version 0.24.0                   
INFO[0000] Checking signature                            index=/home/andrew/.arduino15/package_index.json signatureFile=/home/andrew/.arduino15/package_index.json.sig trusted=true
INFO[0000] Checking if CLI is Bundled into the IDE      
INFO[0000] Loading hardware from: /home/andrew/.arduino15/packages 
INFO[0000] Loading package builtin from: /home/andrew/.arduino15/packages/builtin 
INFO[0000] Checking existence of 'tools' path: /home/andrew/.arduino15/packages/builtin/tools 
INFO[0000] Loading tools from dir: /home/andrew/.arduino15/packages/builtin/tools 
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0000] Loaded tool                                   tool="builtin:mdns-discovery@1.0.5"
INFO[0000] Loaded tool                                   tool="builtin:serial-discovery@1.3.2"
INFO[0000] Loaded tool                                   tool="builtin:serial-monitor@0.9.1"
INFO[0003] Adding libraries dir                          dir=/home/andrew/Arduino/libraries location=user
INFO[0003] Executing `arduino-cli board list`           
INFO[0003] starting discovery builtin:mdns-discovery process 
INFO[0003] starting discovery builtin:serial-discovery process 
INFO[0003] started discovery builtin:serial-discovery process 
INFO[0003] sending command HELLO 1 "arduino-cli 0.24.0" to discovery builtin:serial-discovery 
INFO[0003] sending command QUIT to discovery builtin:mdns-discovery 
ERRO[0003] Quitting discovery builtin:mdns-discovery: EOF 
INFO[0003] killing discovery builtin:mdns-discovery process 
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xb8e30]

goroutine 6 [running]:
os.(*Process).signal(0x0, {0xb8e5a8, 0x102018c})
	/usr/local/go/src/os/exec_unix.go:64 +0x38
os.(*Process).Signal(...)
	/usr/local/go/src/os/exec.go:138
os.(*Process).kill(...)
	/usr/local/go/src/os/exec_posix.go:68
os.(*Process).Kill(...)
	/usr/local/go/src/os/exec.go:123
github.com/arduino/arduino-cli/executils.(*Process).Kill(...)
	/home/build/executils/process.go:120
github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).killProcess(0x3835180)
	/home/build/arduino/discovery/discovery.go:268 +0xdc
github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).Quit(0x3835180)
	/home/build/arduino/discovery/discovery.go:381 +0x104
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).remove(0x366c050, {0x94b78e, 0x16})
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:82 +0x94
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).RunAll.func1(0x3835180)
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:131 +0x78
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize.func1(0x797a2d0, 0x795a410, 0x79f2000, 0x3835180)
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:101 +0x50
created by github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:99 +0x1ac
```

andrewintwdev, 2022-06-28:
> ### Describe the problem
> Raspberry pi 4 | Ubuntu 20.04 LTS ARCH=ARMv7 OS=Linux
> 
> Installed via: curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh -s nightly-latest
> 
> Running: `arduino-cli board list`
> 
> Returns:
> 
> ```
> panic: runtime error: invalid memory address or nil pointer dereference
> [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xb8e30]
> 
> goroutine 11 [running]:
> os.(*Process).signal(0x0, {0xbb4280, 0x106018c})
> 	/usr/local/go/src/os/exec_unix.go:64 +0x38
> os.(*Process).Signal(...)
> 	/usr/local/go/src/os/exec.go:138
> os.(*Process).kill(...)
> 	/usr/local/go/src/os/exec_posix.go:68
> os.(*Process).Kill(...)
> 	/usr/local/go/src/os/exec.go:123
> github.com/arduino/arduino-cli/executils.(*Process).Kill(...)
> 	/home/build/executils/process.go:120
> github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).killProcess(0x2010e80)
> 	/home/build/arduino/discovery/discovery.go:268 +0xdc
> github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).Quit(0x2010e80)
> 	/home/build/arduino/discovery/discovery.go:381 +0x104
> github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).remove(0x1eb2130, {0x96f714, 0x16})
> 	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:82 +0x94
> github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).RunAll.func1(0x2010e80)
> 	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:131 +0x78
> github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize.func1(0x5c0aca0, 0x5b27d70, 0x465a8c0, 0x2010e80)
> 	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:101 +0x50
> created by github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize
> 	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:99 `+0x1ac`
> ```
> 
> ### To reproduce
> arduino-cli board list
> 
> ### Expected behavior
> Return board detected
> 
> ### Arduino CLI version
> arduino-cli Version: nightly-20220618 Commit: [64bc4dc](https://github.com/arduino/arduino-cli/commit/64bc4dcecc14c65a0cf5ecf86b8c89f286b01742) Date: 2022-06-18T01:37:19Z
> 
> ### Operating system
> Linux
> 
> ### Operating system version
> 20.04LTS
> 
> ### Additional context
> _No response_
> 
> ### Issue checklist
> * [x]  I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
> * [x]  I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
> * [x]  My report contains all necessary details
> 
> UPDATE: 0.8.0 version works

I found the root cause, the .arduino15/packages/builtin/tools/mdns-discovery/1.0.5/mdns-discovery needs /lib/ld-linux.so.3, but in my target system, the name is /lib/ld-linux-armhf.so.3. So you need to create a symbolic link.

```
$ sudo ln -s /lib/ld-linux-armhf.so.3 /lib/ld-linux.so.3
$ ls -l /lib/ld-linux.so.3 
lrwxrwxrwx 1 root root 24 Jun 27 23:52 /lib/ld-linux.so.3 -> /lib/ld-linux-armhf.so.3

$ arduino-cli board list
Port         Protocol Type              Board Name FQBN Core
/dev/ttyUSB0 serial   Serial Port (USB) Unknown    
```

Enver-Yilmaz, 2022-07-08:
I'm getting the same error with an M1 Mac without Rosetta 2 installed. I'm trying the native arm64 version of CLI from nightly builds and homebrew installation, and both throw the same error.
```
% arduino-cli board list                     
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x2 addr=0x0 pc=0x100980300]

goroutine 8 [running]:
os.(*Process).signal(0x0, {0x101509060, 0x101a94a78})
	/usr/local/go/src/os/exec_unix.go:64 +0x30
os.(*Process).Signal(...)
	/usr/local/go/src/os/exec.go:138
os.(*Process).kill(...)
	/usr/local/go/src/os/exec_posix.go:68
os.(*Process).Kill(...)
	/usr/local/go/src/os/exec.go:123
github.com/arduino/arduino-cli/executils.(*Process).Kill(...)
	/home/build/executils/process.go:120
github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).killProcess(0x140001d9580)
	/home/build/arduino/discovery/discovery.go:268 +0xd4
github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).Quit(0x140001d9580)
	/home/build/arduino/discovery/discovery.go:381 +0x10c
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).remove(0x14000436a50, {0x1010df9da, 0x16})
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:82 +0x144
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).RunAll.func1(0x140001d9580)
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:131 +0x80
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize.func1(0x140041c9760, 0x140041f8600, 0x1400349cd80, 0x140001d9580)
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:101 +0x58
created by github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize
	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:99 
```

SadDarkAngel, 2022-07-11:
> > ### Describe the problem
> > Raspberry pi 4 | Ubuntu 20.04 LTS ARCH=ARMv7 OS=Linux
> > Installed via: curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh -s nightly-latest
> > Running: `arduino-cli board list`
> > Returns:
> > ```
> > panic: runtime error: invalid memory address or nil pointer dereference
> > [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xb8e30]
> > 
> > goroutine 11 [running]:
> > os.(*Process).signal(0x0, {0xbb4280, 0x106018c})
> > 	/usr/local/go/src/os/exec_unix.go:64 +0x38
> > os.(*Process).Signal(...)
> > 	/usr/local/go/src/os/exec.go:138
> > os.(*Process).kill(...)
> > 	/usr/local/go/src/os/exec_posix.go:68
> > os.(*Process).Kill(...)
> > 	/usr/local/go/src/os/exec.go:123
> > github.com/arduino/arduino-cli/executils.(*Process).Kill(...)
> > 	/home/build/executils/process.go:120
> > github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).killProcess(0x2010e80)
> > 	/home/build/arduino/discovery/discovery.go:268 +0xdc
> > github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).Quit(0x2010e80)
> > 	/home/build/arduino/discovery/discovery.go:381 +0x104
> > github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).remove(0x1eb2130, {0x96f714, 0x16})
> > 	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:82 +0x94
> > github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).RunAll.func1(0x2010e80)
> > 	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:131 +0x78
> > github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize.func1(0x5c0aca0, 0x5b27d70, 0x465a8c0, 0x2010e80)
> > 	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:101 +0x50
> > created by github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize
> > 	/home/build/arduino/discovery/discoverymanager/discoverymanager.go:99 `+0x1ac`
> > ```
> 
> I found the root cause, the `.arduino15/packages/builtin/tools/mdns-discovery/1.0.5/mdns-discovery` needs `/lib/ld-linux.so.3`, but in my target system, the name is `/lib/ld-linux-armhf.so.3`. So you need to create a symbolic link.
> 
> ```
> $ sudo ln -s /lib/ld-linux-armhf.so.3 /lib/ld-linux.so.3
> $ ls -l /lib/ld-linux.so.3 
> lrwxrwxrwx 1 root root 24 Jun 27 23:52 /lib/ld-linux.so.3 -> /lib/ld-linux-armhf.so.3
> 
> $ arduino-cli board list
> Port         Protocol Type              Board Name FQBN Core
> /dev/ttyUSB0 serial   Serial Port (USB) Unknown    
> ```

it’s helps me to (on Orange Pi Zero)

umbynos, 2023-03-17:
This should have been fixed by https://github.com/arduino/mdns-discovery/pull/47#issuecomment-1474122336

umbynos, 2023-03-17:
This is the output when I install the CLI and I run board list:
```
root@d32f9de3786d:/# curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh -s nightly-latest
Installing in //bin
ARCH=ARMv7
OS=Linux
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/nightly/arduino-cli_nightly-latest_Linux_ARMv7.tar.gz
arduino-cli  Version: nightly-20230317 Commit: d5eb0b7 Date: 2023-03-17T01:33:40Z installed successfully in //bin
root@d32f9de3786d:/# arduino-cli version
arduino-cli  Version: nightly-20230317 Commit: d5eb0b7 Date: 2023-03-17T01:33:40Z
root@d32f9de3786d:/# arduino-cli board list -v
INFO[0000] Config file not found, using default values  
INFO[0000] arduino-cli version nightly-20230317         
INFO[0000] Updating libraries index                     
INFO[0000] Starting download                             url="https://downloads.arduino.cc/libraries/library_index.tar.bz2"
Downloading index: library_index.tar.bz2 downloaded                                                                                                                         
INFO[0003] Updating index                                url="https://downloads.arduino.cc/packages/package_index.tar.bz2"
INFO[0003] Starting download                             url="https://downloads.arduino.cc/packages/package_index.tar.bz2"
Downloading index: package_index.tar.bz2 downloaded                                                                                                                         
INFO[0003] Loading hardware from: /root/.arduino15/packages 
INFO[0003] Checking signature                            index=/root/.arduino15/package_index.json signatureFile=/root/.arduino15/package_index.json.sig trusted=true
Downloading missing tool builtin:ctags@5.8-arduino11...
INFO[0003] Starting download                             url="https://downloads.arduino.cc/tools/ctags-5.8-arduino11-pm-armv6-linux-gnueabihf.tar.bz2"
builtin:ctags@5.8-arduino11 downloaded                                                                                                                                      
INFO[0004] Installing tool                               Tool="builtin:ctags@5.8-arduino11"
Installing builtin:ctags@5.8-arduino11...
INFO[0004] Skipping tool configuration.                  Tool="builtin:ctags@5.8-arduino11"
Skipping tool configuration....
INFO[0004] Tool installed                                Tool="builtin:ctags@5.8-arduino11"
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.4.0...
INFO[0004] Starting download                             url="https://downloads.arduino.cc/discovery/serial-discovery/serial-discovery_v1.4.0_Linux_ARMv6.tar.gz"
builtin:serial-discovery@1.4.0 downloaded                                                                                                                                   
INFO[0005] Installing tool                               Tool="builtin:serial-discovery@1.4.0"
Installing builtin:serial-discovery@1.4.0...
INFO[0005] Skipping tool configuration.                  Tool="builtin:serial-discovery@1.4.0"
Skipping tool configuration....
INFO[0005] Tool installed                                Tool="builtin:serial-discovery@1.4.0"
builtin:serial-discovery@1.4.0 installed
Downloading missing tool builtin:mdns-discovery@1.0.8...
INFO[0005] Starting download                             url="https://downloads.arduino.cc/discovery/mdns-discovery/mdns-discovery_v1.0.8_Linux_ARMv6.tar.gz"
builtin:mdns-discovery@1.0.8 downloaded                                                                                                                                     
INFO[0006] Installing tool                               Tool="builtin:mdns-discovery@1.0.8"
Installing builtin:mdns-discovery@1.0.8...
INFO[0006] Skipping tool configuration.                  Tool="builtin:mdns-discovery@1.0.8"
Skipping tool configuration....
INFO[0006] Tool installed                                Tool="builtin:mdns-discovery@1.0.8"
builtin:mdns-discovery@1.0.8 installed
Downloading missing tool builtin:serial-monitor@0.13.0...
INFO[0006] Starting download                             url="https://downloads.arduino.cc/monitor/serial-monitor/serial-monitor_v0.13.0_Linux_ARMv6.tar.gz"
builtin:serial-monitor@0.13.0 downloaded                                                                                                                                    
INFO[0007] Installing tool                               Tool="builtin:serial-monitor@0.13.0"
Installing builtin:serial-monitor@0.13.0...
INFO[0007] Skipping tool configuration.                  Tool="builtin:serial-monitor@0.13.0"
Skipping tool configuration....
INFO[0007] Tool installed                                Tool="builtin:serial-monitor@0.13.0"
builtin:serial-monitor@0.13.0 installed
INFO[0007] Loading tools from dir: /root/.arduino15/packages/builtin/tools 
INFO[0007] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0007] Loaded tool                                   tool="builtin:mdns-discovery@1.0.8"
INFO[0007] Loaded tool                                   tool="builtin:serial-discovery@1.4.0"
INFO[0007] Loaded tool                                   tool="builtin:serial-monitor@0.13.0"
INFO[0007] Loading libraries index file                  index=/root/.arduino15/library_index.json
[... cut ...]
Error starting discovery: discovery builtin:mdns-discovery process not started: calling HELLO: EOF
WARN[0009] Error starting discovery: discovery builtin:mdns-discovery process not started: calling HELLO: EOF 
No boards found.
```
It does not start correctly the `mdns-discovery` tool (because it's still using the v1.0.8 which has the problem explained by @andrewintw [here](https://github.com/arduino/arduino-cli/issues/1775#issuecomment-1168168582)), but definitely it's not panicking. So I would close this one.