arduino/arduino-cli #1785: Preprocessing bug (probably).

jfjlaros, 2022-06-27:


facchinm, 2022-06-27:
@per1234 done :wink: 
@jfjlaros it's indeed a nasty bug, mostly due to the fact that `ctags` is not very template friendly, so we had to create special cases and probably multiline is not well tested. Thanks for the PoC btw

jfjlaros, 2022-07-05:
Here is an other one. The following function definition

```cpp
template <size_t N>
void func(double const (&)[N]) {}
```
gives rise to these errors (different than the ones in the previous example)

```txt
arduino-cli compile --fqbn arduino:avr:pro --warnings all --output-dir build \
    --build-property compiler.cpp.extra_flags="-pedantic"
.../x.ino:3:45: error: variable or field 'func' declared void
.../x.ino:3:30: error: expected primary-expression before 'double'
```

while

```cpp
template <size_t N> void func(double const (&)[N]) {}
```

compiles fine.