arduino/arduino-cli #1792: Link fails: some object files are missing

FStefanni, 2022-07-07:
### Describe the problem

Hi,

some times (I have been unable to understand why) the link fails with messages as the following:

```bash
/home/kappa/.arduino15/packages/Heltec-esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
/home/kappa/.arduino15/packages/Heltec-esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
```

As you can see, these are part of the esp32 packages, and I expect them to be compiled and linked successfully.
Please note that since also sometimes links successfully, it seems something strange, maybe related to the timestamps...

Regards

### To reproduce

I am unable to understand how to reproduce it.
When compiling huge programs, sometimes it happen.

### Expected behavior

To compile and link successfully.

### Arduino CLI version

0.23.0

### Operating system

Linux

### Operating system version

Debian Testing

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

FStefanni, 2022-07-07:
Hi,

after few investigation, I can confirm:

1. It fails also with the minimal standard Blink sketch
2. This is the output of two consecutive runs:
```bash
kappa@kappa-debian:~/temp/Blink$ arduino-cli compile --fqbn esp32:esp32:esp32 --output-dir . ./Blink.ino
Sketch uses 219005 bytes (16%) of program storage space. Maximum is 1310720 bytes.
Global variables use 16088 bytes (4%) of dynamic memory, leaving 311592 bytes for local variables. Maximum is 327680 bytes.

kappa@kappa-debian:~/temp/Blink$ arduino-cli compile --fqbn esp32:esp32:esp32 --output-dir . ./Blink.ino
/home/kappa/.arduino15/packages/Heltec-esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
/home/kappa/.arduino15/packages/Heltec-esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
collect2: error: ld returned 1 exit status

Error during build: exit status 1
```
3. After some consecutive fails, it re-compiles again successfully for one time, and then fails again and so on.
4. When it fails, it uses the wrong package (the Heltec), when it succeeds, it uses the correct one (esp32)
5. I have done few attempts with older versions, but it fails nevertheless (I updated the versions at the top of this issue)

Regards

cmaglie, 2023-04-27:
Hi @FStefanni, thanks for the bug report!

Could you re-test with the latest Arduino CLI? we merged some fixes on the tools selection subroutines.
If the issue is still present, may you give a link to the Heltec package to try?

FStefanni, 2023-04-27:
Hi,

I have tested with version 0.32.2: I compiled the Blink example for ten consecutive times, and none failed.
I have performed also the same test with the Heltec board, and it also has gone fine.

So maybe some of the fixes have been successful also for this issue.

So I close this issue, since it seems fixed.
Regards.

closed 2023-04-27 by FStefanni