arduino/arduino-cli #1816: using Arduino-cli with docker to flash the board fails

VojislavM, 2022-07-27:
### Describe the problem

Currently I am using Arduino-cli to flash the device connected to the USB. Everything is running on the Raspberry Pi. The devices are exposed to docker container with additional parameter in the command:
```
--privileged -v /dev:/dev
```
when trying to run `arduino-cli board list` I get an error:
```
root@8155728bcd2d:/opt/app# arduino-cli board list
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xb8e30]

goroutine 49 [running]:
os.(*Process).signal(0x0, {0xba56b0, 0x104018c})
        /usr/local/go/src/os/exec_unix.go:64 +0x38
os.(*Process).Signal(...)
        /usr/local/go/src/os/exec.go:138
os.(*Process).kill(...)
        /usr/local/go/src/os/exec_posix.go:68
os.(*Process).Kill(...)
        /usr/local/go/src/os/exec.go:123
github.com/arduino/arduino-cli/executils.(*Process).Kill(...)
        /home/build/executils/process.go:120
github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).killProcess(0x2c99c80)
        /home/build/arduino/discovery/discovery.go:268 +0xdc
github.com/arduino/arduino-cli/arduino/discovery.(*PluggableDiscovery).Quit(0x2c99c80)
        /home/build/arduino/discovery/discovery.go:381 +0x104
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).remove(0x2ee20d0, {0x95f016, 0x16})
        /home/build/arduino/discovery/discoverymanager/discoverymanager.go:82 +0x94
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).RunAll.func1(0x2c99c80)
        /home/build/arduino/discovery/discoverymanager/discoverymanager.go:131 +0x78
github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize.func1(0x72147d0, 0x71eca68, 0x56ad240, 0x2c99c80)
        /home/build/arduino/discovery/discoverymanager/discoverymanager.go:101 +0x50
created by github.com/arduino/arduino-cli/arduino/discovery/discoverymanager.(*DiscoveryManager).parallelize
        /home/build/arduino/discovery/discoverymanager/discoverymanager.go:99 +0x1ac
root@8155728bcd2d:/opt/app# 
```

I do not have this issue if I am using Arduino-cli version **0.18.0** 

and I checked the ports are exposed in the container and can be accessed with screen or other terminal tools.

### To reproduce

to reproduce you can build the minimal docker image:
```Docker
FROM python:3.9
WORKDIR /opt/app
COPY . /opt/app
ENV PATH="${PATH}:/home/$USER/bin"
RUN apt-get update && apt-get install -y \
curl
RUN curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=/usr/bin sh;
RUN arduino-cli core update index
RUN arduino-cli core install arduino:mbed@1.1.6
```
and then build the docker image with command:
```
docker build -t test_flash_image:1.0.0 .
```
and then run the environment:
```
docker run -it -v /dev:/dev --privileged {your_image_id} bash
```
after getting the image console run the Arduino-cli:
```
arduino-cli board list
```

### Expected behavior

behaviour should be same as Arduino-cli working without docker natively on the host OS.

### Arduino CLI version

0.25.1

### Operating system

Linux

### Operating system version

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-07-27:
Hi @VojislavM. Does the issue still occur if you run the `arduino-cli board list` command without using a container?

The reason I ask is because we already have multiple reports of the same problem occurring on Linux ARM hosts even without a container (https://github.com/arduino/arduino-cli/issues/1775), which makes me think the container is not a relevant factor to this fault.

VojislavM, 2022-07-27:
No, everything works when i use it without the container

cmaglie, 2022-07-27:
Is this comment relevant to you? 
https://github.com/arduino/arduino-cli/issues/1775#issuecomment-1168168582

> I found the root cause, the .arduino15/packages/builtin/tools/mdns-discovery/1.0.5/mdns-discovery needs /lib/ld-linux.so.3, but in my target system, the name is /lib/ld-linux-armhf.so.3. So you need to create a symbolic link.
> 
> ```
> $ sudo ln -s /lib/ld-linux-armhf.so.3 /lib/ld-linux.so.3
> $ ls -l /lib/ld-linux.so.3 
> lrwxrwxrwx 1 root root 24 Jun 27 23:52 /lib/ld-linux.so.3 -> /lib/ld-linux-armhf.so.3
> 
> $ arduino-cli board list
> Port         Protocol Type              Board Name FQBN Core
> /dev/ttyUSB0 serial   Serial Port (USB) Unknown    
> ```

Could it be that the container limit access to `/lib/ld-linux-armhf.so.3`?

cmaglie, 2023-04-28:
Hi @VojislavM, we have merged many fixes on discovery handling, could you retry with the latest Arduino CLI?
In case the issue is still present please copy also the updated stack trace.

spivakdm, 2023-05-14:
I have the same issue with running arduino-cli in docker container on raspberry pi 4. Arduino-cli ver. 0.32.2

arduino-cli board list works fine for me, but trying to access serial monitor 
`arduino-cli monitor -p '/dev/ttyACM0' `
returns this error: 
Error getting port settings details: Port monitor error: fork/exec /root/.arduino15/packages/builtin/tools/serial-monitor/0.13.0/serial-monitor: no such file or directory

Similar error when trying to upload the code 
`arduino-cli compile -u -b arduino:avr:uno /app/sketch_templates/serial.ino `

Error during build: fork/exec /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++: no such file or directory

Those files definitely exist , checked that, there is no problem with the access also. Core driver is installed also.
Any ideas how to fix it?

cmaglie, 2023-05-15:
Hi @spivakdm,

please paste the output of the following command:

```
ldd /root/.arduino15/packages/builtin/tools/serial-monitor/0.13.0/serial-monitor
```

and may you also try to run `serial-monitor` directly:

```
/root/.arduino15/packages/builtin/tools/serial-monitor/0.13.0/serial-monitor
```

I want to see if `serial-monitor` runs without problems or is something related to the container.
If `serial-monitor` runs correctly it should print nothing and just wait for input, in that case just hit CTRL-C to exit, otherwise, please paste the error here.

spivakdm, 2023-05-15:
Hi @cmaglie , 

I get this error when executing command
```
root@04de19a5362a:/app# ldd /root/.arduino15/packages/builtin/tools/serial-monitor/0.13.0/serial-monitor
        not a dynamic executable
```

I guess I am missing some dependencies inside the container. Some gcc compiler? I wonder which one, is there a way to fin

cmaglie, 2023-05-15:
There is the avr-g++ compiler to try too:

```
ldd /root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++
```

hopefully, this may give some hints.

spivakdm, 2023-05-15:
Returns the same error, although file is there. 
```
/root/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++: 
No such file or directory
```

I have also tried running it on Raspberry pi outside of the container. And it recognizes file as executable, is able to launch it and it is working as you described. 
Thing is Raspberry pi has 64bit OS and there is 32bit Raspbian OS inside the container. I believe there might be some problems with 32bit architecture, maybe the wrong version is installed though I used the shell script to iinstall it. I will try to explore that.

VojislavM, 2023-05-29:
just to confirm, using CLI from docker on under arm host works fine for me for some time now. Fell free to close the issue.

closed 2023-06-12 by umbynos