arduino/arduino-cli #1821: Allow downloading lib_index and signature in a single tar.bz2 archive

umbynos, 2022-08-03:
Similar to https://github.com/arduino/arduino-cli/pull/1734 but for library index

kittaakos, 2022-08-03:
Originally reported by @kittaakos 

Happens during the IDE2 startup.
```
root ERROR Failed to update indexes Error: 7 PERMISSION_DENIED: Error verifying signature: openpgp: invalid signature: hash tag doesn't match
    at Object.callErrorFromStatus (/Users/a.kitta/dev/git/arduino-ide/node_modules/@grpc/grpc-js/build/src/call.js:31:26)
    at Object.onReceiveStatus (/Users/a.kitta/dev/git/arduino-ide/node_modules/@grpc/grpc-js/build/src/client.js:349:49)
    at Object.onReceiveStatus (/Users/a.kitta/dev/git/arduino-ide/node_modules/@grpc/grpc-js/build/src/client-interceptors.js:328:181)
    at /Users/a.kitta/dev/git/arduino-ide/node_modules/@grpc/grpc-js/build/src/call-stream.js:187:78
    at processTicksAndRejections (node:internal/process/task_queues:78:11)
```

Running these manually fixed the issue:
```
a.kitta@Akoss-MacBook-Pro arduino-ide % ./arduino-ide-extension/build/arduino-cli core update-index --format json
a.kitta@Akoss-MacBook-Pro arduino-ide % ./arduino-ide-extension/build/arduino-cli lib update-index --format json
```

@cmaglie commented:

> from time to time we get this error from the lib update index

> on the CI

> the hypothesis is that the cloudflare cache is not aligned for some millisecond and if you download the signature in that instant the signature do not match

> probably the most reliable solution is to do what we did with the package_index -> create an tar.bz2 archive with both signature and index, so they are downloaded togherter

cmaglie, 2022-10-12:
Fixed by #1908

closed 2022-10-12 by cmaglie