arduino/arduino-cli #1849: Support of caret and tilde in version contraints

FStefanni, 2022-08-25:
### Describe the request

Hi,

many package manager support as version constraints also caret `^` and tilde `~`:

* `^2.3.4` allows changes that do not modify the major digit (e.g.: `2.9.1` is ok, `3.0.0` is ko)
* `~2.3.4` allows for changes that increase only the patch digit (e.g.: `2.3.5` is ok, `2.4.0` is ko)

(specific semantics is up to the package manager, e.g. [npm](https://docs.npmjs.com/cli/v6/using-npm/semver#tilde-ranges-123-12-1) has slightly more complex semantics)

This would allow to specify library dependencies in a more specific way, supporting [the semver standard](https://semver.org/).
Instead, as far as I know, currently constraints only supports generic "greater-than" et similar.

If accepted this proposal, also the package installation tools should be updated accordingly (e.g. `arduino-cli`).

Regards

### Describe the current behavior

As far as I know, currently constraints only supports generic "greater-than" et similar, preventing to apply the semver standard.

### Arduino CLI version

all

### Operating system

N/A

### Operating system version

all

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-08-26:
Hi @FStefanni. Thanks for your request.

The version constraints syntax is provided to Arduino CLI by [the `go.bug.st/relaxed-semver` Go module](https://github.com/bugst/relaxed-semver). So any requests must be submitted to that repository. Any syntax provided by that module will also eventually be used by the Arduino library system.

You'll be glad to know that the `^` operator has already been added to that module: https://github.com/bugst/relaxed-semver/pull/8. This means it is only a matter of time for that change to be released and then finally pulled into Arduino CLI when the a dependency version is bumped.

Maybe `^` alone is sufficient?

There is also a proposal for a change in the way constraints without operator are treated that you might be interested in reviewing: https://github.com/bugst/relaxed-semver/pull/10. That PR also provides the missing documentation for the `^` operator.

>As far as I know, currently constraints only supports generic "greater-than" et similar

The constraints syntax of the module is documented in its readme:

https://github.com/bugst/relaxed-semver#version-constraints

I felt that the way they are documented there, although completely appropriate for those using the modules directly, would not be very clear to those using them indirectly in Arduino library metadata, so I added equivalent documentation for the syntax as used in this specific application in the Arduino Library Specification:

https://arduino.github.io/arduino-cli/dev/library-specification/#version-constraints

> preventing to apply the semver standard.

I think this statement might be a bit misleading.

Even though someone reading the documentation of **npm** and the ["**semver**" package](https://github.com/npm/node-semver) (which, despite the name, is an independent project from the semver specification) might be given the impression otherwise, [the real semver specification](https://semver.org/) doesn't say anything at all about constraints.

So it isn't accurate to say the absence of operators established completely separate from the specification does not mean that our syntax is not compliant with the semver specification.

closed 2022-08-26 by per1234

FStefanni, 2022-08-26:
Hi,

thank you for the very complete reply. So I hope that caret and tilde will be available asap.

For the semver application, I mean the following. The main point of having samever is to know that a range dependency versions is compatible out of the box. This is important for example, to be able to update dependencies which provide bug fixes. But specify `>=` as a dependency constraint actually defeat this purpose, because allows also breaking changes, basically forcing to specify exact version matches and therefore, to perform a new release just to switch to a newer dependency version. 
So, even if I agree to what you said about semver (what you said is totally true), imho this is a huge issue in the convenience of semver applying (as using something very powerful in a very limited way). 

Regards