arduino/arduino-cli #1854: `task build` failing with `i18n/detect.go:24:9: undefined: getLocaleIdentifier`

duncan-bayne, 2022-08-29:
### Describe the problem

When I try to build arduino-cli with `task build`, I receive the following error:

```
$ task build
task: [go:build] go build -v -ldflags ' -X github.com/arduino/arduino-cli/version.versionString=git-snapshot -X github.com/arduino/arduino-cli/version.commit=aa41d72e -X github.com/arduino/arduino-cli/version.date=2022-08-29T12:47:49Z '
github.com/arduino/arduino-cli/i18n
# github.com/arduino/arduino-cli/i18n
i18n/detect.go:24:9: undefined: getLocaleIdentifier
task: Failed to run task "go:build": exit status 2
```

### To reproduce

1. `git clone https://github.com/arduino/arduino-cli/`
2. `cd arduino-cli`
3. `task build`

### Expected behavior

I'd expect arduino-cli to compile without errors.

### Arduino CLI version

aa41d72ee6fd422980a2ba43310c269bf0ae2e1f

### Operating system

N/A

### Operating system version

FreeBSD 13.1-RELEASE GENERIC amd64

### Additional context

* FreeBSD 13.1-RELEASE GENERIC amd64
* Task version: v3.14.1 (h1:myTmEIbMbQx2D+g5lJvnbSqX531HmxiaQuefAqz8ISA=)
* go version go1.18.5 freebsd/amd64

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-08-30:
Hi @duncan-bayne. Thanks for your report.

The `github.com/arduino/arduino-cli/i18n.getLocaleIdentifier` function is defined in files which have [build constraints](https://pkg.go.dev/cmd/go#hdr-Build_constraints) via the file name suffix for the following operating systems:

- **macOS**: https://github.com/arduino/arduino-cli/blob/aa41d72ee6fd422980a2ba43310c269bf0ae2e1f/i18n/detect_darwin.go
- **Linux**: https://github.com/arduino/arduino-cli/blob/aa41d72ee6fd422980a2ba43310c269bf0ae2e1f/i18n/detect_linux.go
- **Windows**: https://github.com/arduino/arduino-cli/blob/aa41d72ee6fd422980a2ba43310c269bf0ae2e1f/i18n/detect_windows.go

The file with the suffix matching [the `GOOS` environment variable](https://go.dev/doc/install/source#environment) value will be built. If no file suffix matches `GOOS`, build fails with the error you encountered.

duncan-bayne, 2022-08-30:
Okay thanks - that makes sense.  Would you accept a pull request with the necessary implementation for FreeBSD?

duncan-bayne, 2022-09-02:
Looking into this a bit further, there is a good range of Arduino related ports available in FreeBSD already, including all of the dependencies of arduino-cli and the IDE.  I've tested those, and I can program my Lilypad through the IDE perfectly well.  

A naive approach to get arduino-cli building fails because the dependencies aren't available for download:

```
./arduino-cli upgrade
Downloading missing tool builtin:ctags@5.8-arduino11...
Downloading missing tool builtin:serial-discovery@1.3.2...
Downloading missing tool builtin:mdns-discovery@1.0.6...
Downloading missing tool builtin:serial-monitor@0.9.1...
Error initializing instance: downloading builtin:ctags@5.8-arduino11 tool: Error downloading tool builtin:ctags@5.8-arduino11: no versions available for the current OS
Error initializing instance: downloading builtin:serial-discovery@1.3.2 tool: Error downloading tool builtin:serial-discovery@1.3.2: no versions available for the current OS
Error initializing instance: downloading builtin:mdns-discovery@1.0.6 tool: Error downloading tool builtin:mdns-discovery@1.0.6: no versions available for the current OS
Error initializing instance: downloading builtin:serial-monitor@0.9.1 tool: Error downloading tool builtin:serial-monitor@0.9.1: no versions available for the current OS
Error initializing instance: Error loading hardware platform: reading directory packages/builtin/tools: open packages/builtin/tools: no such file or directory
Error initializing instance: Error loading hardware platform: discovery builtin:serial-discovery not installed
Error initializing instance: Error loading hardware platform: discovery builtin:mdns-discovery not installed
[duncan@duncan-x250 ~/code/arduino-cli]$ 
```

So it looks like maybe the answer is for me to create a port of arduino-cli that in turn depends upon the _existing_ ports?  Possibly after adding FreeBSD support to this library to make the resultant patch smaller.

(Note that this may also address https://github.com/arduino/arduino-cli/issues/1815).

umbynos, 2022-09-05:
Hello @duncan-bayne! Thanks for your interest in this project!
Unfortunately, as you may already know by now, your operative system is not supported at the moment.
The Arduino CLI is not the only piece of software that needs fixing in order to add the support.
As you can see from the output:
- [ctags](https://github.com/arduino/ctags)
- [serial-discovery](https://github.com/arduino/serial-discovery/)
- [mdns-discovery](https://github.com/arduino/mdns-discovery/)
- [serial-monitor](https://github.com/arduino/serial-monitor/)
Needs a proper release for FreeBSD.
Unfortunately, I don't think we have the bandwidth at the moment...
But of course we are open to contributors.

per1234, 2023-03-28:
Closing as fixed by https://github.com/arduino/arduino-cli/pull/2116

The other issue of official builds of the tool dependencies not being available for this host type remains, but that should be tracked in the repositories of those projects instead of here.

There is a related request for official builds of Arduino CLI for FreeBSD here: https://github.com/arduino/arduino-cli/issues/1815

closed 2023-03-28 by per1234