arduino/arduino-cli #1856: `lib upgrade` does not install dependencies

per1234, 2022-08-30:
### Describe the problem

Arduino libraries may have dependencies on other libraries. These can be specified in the [library's metadata](https://arduino.github.io/arduino-cli/dev/library-specification/#library-metadata), which will cause Arduino CLI to install them along with the library.

🐛 The dependencies of a library are not installed when a new version of the dependent library is installed by `arduino-cli lib upgrade`

### To reproduce

```text
$ arduino-cli version
arduino-cli.exe  Version: git-snapshot Commit: 05d14464 Date: 2022-08-30T19:43:20Z

$ arduino-cli lib uninstall MKRWAN
Uninstalling MKRWAN@1.1.0...

$ arduino-cli lib install Arduino_ConnectionHandler@0.3.3
Already installed MKRGSM@1.5.0
Already installed MKRNB@1.5.1
Already installed Arduino_DebugUtils@1.3.0
Already installed WiFi101@0.16.1
Already installed WiFiNINA@1.8.13
Downloading Arduino_ConnectionHandler@0.3.3...
Arduino_ConnectionHandler@0.3.3 already downloaded
Installing Arduino_ConnectionHandler@0.3.3...
Installed Arduino_ConnectionHandler@0.3.3

$ arduino-cli lib deps Arduino_ConnectionHandler@0.3.3
✓ MKRNB 1.5.1 is already installed.
✓ Arduino_ConnectionHandler 0.3.3 is already installed.
✓ Arduino_DebugUtils 1.3.0 is already installed.
✓ WiFi101 0.16.1 is already installed.
✓ WiFiNINA 1.8.13 is already installed.
✓ MKRGSM 1.5.0 is already installed.

$ arduino-cli lib upgrade
Downloading Arduino_ConnectionHandler@0.6.6...
Arduino_ConnectionHandler@0.6.6 already downloaded
Installing Arduino_ConnectionHandler@0.6.6...
Replacing Arduino_ConnectionHandler@0.3.3 with Arduino_ConnectionHandler@0.6.6...
Installed Arduino_ConnectionHandler@0.6.6

$ arduino-cli lib deps Arduino_ConnectionHandler@0.6.6
✓ WiFiNINA 1.8.13 is already installed.
✓ MKRGSM 1.5.0 is already installed.
✓ MKRNB 1.5.1 is already installed.
✕ MKRWAN 1.1.0 must be installed.
✓ Arduino_ConnectionHandler 0.6.6 is already installed.
✓ Arduino_DebugUtils 1.3.0 is already installed.
✓ WiFi101 0.16.1 is already installed.
```

🐛 The new dependency `MKRWAN` (added after the time of the `Arduino_ConnectionHandler@0.3.3` release) was not installed.

### Expected behavior

Install dependencies of libraries upgraded by `arduino-cli lib upgrade`, just as already happens if the library was individually upgraded by running `arduino-cli lib install Arduino_ConnectionHandler`.

### Arduino CLI version

05d14464

### Operating system

Windows

### Operating system version

Windows 10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

KurtE, 2022-10-06:
Not sure if this is the same issue or not?

Selecting to option to update a library and all of its dependencies, did not appear to do anything. 

Yesterday I opened up the daily IDE build I had installed on my Windows machine.  Maybe 1005 or 1004 I think.

It came up with the list of libraries.  There were several of the Adafruit libraries that showed that needed to be updated.
I selected one of them and it said it depended on some other libraries.  With the three options: Install all, install only this one, ...

I selected install all of them and it appeared like nothing happened.  There was zero output in the output window.  I then clicked on the install only this one and it did install that specific one.    I repeated the same thing with a couple of others, and same thing.

I am now running todays build, but I don't have any more of these dependencies to try it with.

per1234, 2022-10-07:
Hi @KurtE. There was a bug in the IDE that caused the "**INSTALL ALL**" and "**CANCEL**" buttons to be swapped on that dialog. It was fixed a few days ago so it should no longer occur when using the latest nightly build. Please let us know if you still experience the problem with the updated version.

More information here: https://github.com/arduino/arduino-ide/pull/1512

PR #1935 (2022-10-21): Fix lib upgrade command not installing new libraries dependencies
## Please check if the PR fulfills these requirements

See [how to contribute](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/)

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug f [... cut]
closed 2022-10-21 by cmaglie