arduino/arduino-cli #1869: arduino-cli USB CDC enable function requirement

yoyojacky, 2022-09-08:
### Describe the request

I want to enable or disable ESP32-C3 module which is from DFRobot,  and it works well under arduino-cli, but once i use  `arduino-cli monitor -p /dev/ttyACM0 ` there is no output,  after searching and discussing with my friends , he said we need to enable USB CDC during flashing the chip,   but i can not find the way to add parameter's about enabling USB CDC. 

### Describe the current behavior

It is works well under arduino-cli, but once i use  `arduino-cli monitor -p /dev/ttyACM0 ` there is no output,  after searching and discussing with my friends , he said we need to enable USB CDC during flashing the chip,   but i can not find the way to add parameter's about enabling USB CDC. 

### Arduino CLI version

0.26.0-> 0.27.1

### Operating system

Linux

### Operating system version

Debian 11 Bullseye 64bit on Raspberry Pi 4B 

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-09-08:
Hi @yoyojacky. Custom board options such as "**USB CDC On Boot**" are set via the FQBN you pass to `arduino-cli` commands via the `--fqbn` flag.

The format of the FQBN is like this:

```text
<vendor ID>:<architecture>:<board ID>[:<menu ID>=<option ID>[,<menu ID>=<option ID>]...]
```

You can learn all the available menu IDs and option IDs for a given board by running [the following command](https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_board_details/):

```text
arduino-cli board details --fqbn <FQBN>
```

(where `<FQBN>` is replaced by the fully qualified board name of the board you are using)

For example if you are using the "**ESP32C3 Dev Module**" board (FQBN: `esp32:esp32:esp32c3`):

```text
$ ./arduino-cli board details -b esp32:esp32:esp32c3
Board name:            ESP32C3 Dev Module
FQBN:                  esp32:esp32:esp32c3
Board version:         2.0.4

Package name:          esp32
Package maintainer:    Espressif Systems
Package URL:           https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json    
Package website:       https://github.com/espressif/arduino-esp32
Package online help:   http://esp32.com

Platform name:         esp32
Platform category:     ESP32
Platform architecture: esp32
Platform URL:          https://github.com/espressif/arduino-esp32/releases/download/2.0.4/esp32-2.0.4.zip
Platform file name:    esp32-2.0.4.zip
Platform size (bytes): 259715595
Platform checksum:     SHA-256:832609d6f4cd0edf4e471f02e30b7f0e1c86fdd1b950990ef40431e656237214

Required tool:         esp32:riscv32-esp-elf-gcc                                            gcc8_4_0-esp-2021r2-patch3

Required tool:         esp32:xtensa-esp32-elf-gcc                                           gcc8_4_0-esp-2021r2-patch3

Required tool:         esp32:xtensa-esp32s2-elf-gcc                                         gcc8_4_0-esp-2021r2-patch3

Required tool:         esp32:xtensa-esp32s3-elf-gcc                                         gcc8_4_0-esp-2021r2-patch3

Required tool:         esp32:esptool_py                                                     3.3.0

Required tool:         esp32:mkspiffs                                                       0.2.3

Required tool:         esp32:mklittlefs                                                     3.0.0-gnu12-dc7f933

Option:                USB CDC On Boot                                                                                 CDCOnBoot
                       Disabled                                                             ✔                          CDCOnBoot=default
                       Enabled                                                                                         CDCOnBoot=cdc
Option:                Partition Scheme                                                                                PartitionScheme
                       Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)                     ✔                          PartitionScheme=default
                       Default 4MB with ffat (1.2MB APP/1.5MB FATFS)                                                   PartitionScheme=defaultffat
                       8M Flash (3MB APP/1.5MB FAT)                                                                    PartitionScheme=default_8MB
                       Minimal (1.3MB APP/700KB SPIFFS)                                                                PartitionScheme=minimal
                       No OTA (2MB APP/2MB SPIFFS)                                                                     PartitionScheme=no_ota
                       No OTA (1MB APP/3MB SPIFFS)                                                                     PartitionScheme=noota_3g
                       No OTA (2MB APP/2MB FATFS)                                                                      PartitionScheme=noota_ffat
                       No OTA (1MB APP/3MB FATFS)                                                                      PartitionScheme=noota_3gffat
                       Huge APP (3MB No OTA/1MB SPIFFS)                                                                PartitionScheme=huge_app
                       Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)                                                PartitionScheme=min_spiffs
                       16M Flash (2MB APP/12.5MB FAT)                                                                  PartitionScheme=fatflash
                       16M Flash (3MB APP/9MB FATFS)                                                                   PartitionScheme=app3M_fat9M_16MB
                       RainMaker                                                                                       PartitionScheme=rainmaker
Option:                CPU Frequency                                                                                   CPUFreq
[... 33 lines cut ...]
```

Here we can see that the ID for the "**USB CDC On Boot**" menu is `CDCOnBoot` and the ID for the "**Enabled**" option is `cdc`.

So the full FQBN is:

```text
esp32:esp32:esp32c3:CDCOnBoot=cdc
```

If you have any additional questions about using Arduino CLI, feel free to post on the dedicated Arduino Forum section here:

https://forum.arduino.cc/c/software/arduino-cli/89

More information about custom board options here:

https://arduino.github.io/arduino-cli/latest/platform-specification/#custom-board-options

And about setting them via the FQBN here:

https://arduino.github.io/arduino-cli/latest/FAQ/#how-to-set-multiple-board-options

We are tracking the need to better document the FQBN format here:

https://github.com/arduino/arduino-cli/issues/355

yoyojacky, 2022-09-08:
> Hi @yoyojacky. Custom board options such as "**USB CDC On Boot**" are set via the FQBN you pass to `arduino-cli` commands via the `--fqbn` flag.
> 
> The format of the FQBN is like this:
> 
> ```
> <vendor ID>:<architecture>:<board ID>[:<menu ID>=<option ID>[,<menu ID>=<option ID>]...]
> ```
> 
> You can learn all the available menu IDs and option IDs for a given board by running [the following command](https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_board_details/):
> 
> ```
> arduino-cli board details --fqbn <FQBN>
> ```
> 
> (where `<FQBN>` is replaced by the fully qualified board name of the board you are using)
> 
> For example if you are using the "**ESP32C3 Dev Module**" board (FQBN: `esp32:esp32:esp32c3`):
> 
> ```
> $ ./arduino-cli board details -b esp32:esp32:esp32c3
> Board name:            ESP32C3 Dev Module
> FQBN:                  esp32:esp32:esp32c3
> Board version:         2.0.4
> 
> Package name:          esp32
> Package maintainer:    Espressif Systems
> Package URL:           https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json    
> Package website:       https://github.com/espressif/arduino-esp32
> Package online help:   http://esp32.com
> 
> Platform name:         esp32
> Platform category:     ESP32
> Platform architecture: esp32
> Platform URL:          https://github.com/espressif/arduino-esp32/releases/download/2.0.4/esp32-2.0.4.zip
> Platform file name:    esp32-2.0.4.zip
> Platform size (bytes): 259715595
> Platform checksum:     SHA-256:832609d6f4cd0edf4e471f02e30b7f0e1c86fdd1b950990ef40431e656237214
> 
> Required tool:         esp32:riscv32-esp-elf-gcc                                            gcc8_4_0-esp-2021r2-patch3
> 
> Required tool:         esp32:xtensa-esp32-elf-gcc                                           gcc8_4_0-esp-2021r2-patch3
> 
> Required tool:         esp32:xtensa-esp32s2-elf-gcc                                         gcc8_4_0-esp-2021r2-patch3
> 
> Required tool:         esp32:xtensa-esp32s3-elf-gcc                                         gcc8_4_0-esp-2021r2-patch3
> 
> Required tool:         esp32:esptool_py                                                     3.3.0
> 
> Required tool:         esp32:mkspiffs                                                       0.2.3
> 
> Required tool:         esp32:mklittlefs                                                     3.0.0-gnu12-dc7f933
> 
> Option:                USB CDC On Boot                                                                                 CDCOnBoot
>                        Disabled                                                             ✔                          CDCOnBoot=default
>                        Enabled                                                                                         CDCOnBoot=cdc
> Option:                Partition Scheme                                                                                PartitionScheme
>                        Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)                     ✔                          PartitionScheme=default
>                        Default 4MB with ffat (1.2MB APP/1.5MB FATFS)                                                   PartitionScheme=defaultffat
>                        8M Flash (3MB APP/1.5MB FAT)                                                                    PartitionScheme=default_8MB
>                        Minimal (1.3MB APP/700KB SPIFFS)                                                                PartitionScheme=minimal
>                        No OTA (2MB APP/2MB SPIFFS)                                                                     PartitionScheme=no_ota
>                        No OTA (1MB APP/3MB SPIFFS)                                                                     PartitionScheme=noota_3g
>                        No OTA (2MB APP/2MB FATFS)                                                                      PartitionScheme=noota_ffat
>                        No OTA (1MB APP/3MB FATFS)                                                                      PartitionScheme=noota_3gffat
>                        Huge APP (3MB No OTA/1MB SPIFFS)                                                                PartitionScheme=huge_app
>                        Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)                                                PartitionScheme=min_spiffs
>                        16M Flash (2MB APP/12.5MB FAT)                                                                  PartitionScheme=fatflash
>                        16M Flash (3MB APP/9MB FATFS)                                                                   PartitionScheme=app3M_fat9M_16MB
>                        RainMaker                                                                                       PartitionScheme=rainmaker
> Option:                CPU Frequency                                                                                   CPUFreq
[... 35 lines cut ...]
> Here we can see that the ID for the "**USB CDC On Boot**" menu is `CDCOnBoot` and the ID for the "**Enabled**" option is `cdc`.
> 
> So the full FQBN is:
> 
> ```
> esp32:esp32:esp32c3:CDCOnBoot=cdc
> ```
> 
> If you have any additional questions about using Arduino CLI, feel free to post on the dedicated Arduino Forum section here:
> 
> https://forum.arduino.cc/c/software/arduino-cli/89
> 
> More information about custom board options here:
> 
> https://arduino.github.io/arduino-cli/latest/platform-specification/#custom-board-options
> 
> And about setting them via the FQBN here:
> 
> https://arduino.github.io/arduino-cli/latest/FAQ/#how-to-set-multiple-board-options
> 
> We are tracking the need to better document the FQBN format here:
> 
> #355

Thank you very much!!!  It works!!!  you are my god.... thank you again.

closed 2022-09-08 by yoyojacky