arduino/arduino-cli #1884: gRPC output does not support animated progress bars

per1234, 2022-09-20:
### Describe the problem

Some command line tools generate an animated progress bar by using a carriage return without newline to repeatedly overwrite the same line.

🐛 This does not work when using the Arduino CLI gRPC interface. The response message does not contain the carriage returns.

### To reproduce

1. Upload a sketch via the Arduino CLI command line interface to a board that uses an upload tool that produces an animated progress bar.
   🙂 There is a nice looking animated progress bar:
   [image]
1. Do the same using Arduino IDE 1.8.19.
   🙂 There is a nice looking animated progress bar:
   [image]
1. Do the same using Arduino IDE 2.x
   🐛 The progress bar is not animated:
   [image]

### Expected behavior

Animated progress bars are supported when Arduino CLI is used via the gRPC interface.

### Arduino CLI version

87d2efa1

### Operating system

Windows, Ubuntu

### Operating system version

Windows 10, Ubuntu 20.04

### Additional context

Support was added in Arduino IDE 1.x by https://github.com/arduino/Arduino/pull/9954

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2022-09-20:
This is a problem in the console text area of the IDE becuase running `arduino-cli` from the command line shows the expected progress bar: this means that the standrad-out stream from `avrdude` is correctly sent through gRPC.

We had this problem [before in the Arduino IDE 1.8.x](https://github.com/arduino/Arduino/pull/9954), it basically boils down to the text area not correctly handling the [`CR` (Carriage Return) character](https://developer.mozilla.org/en-US/docs/Glossary/CRLF?retiredLocale=it):

> CR and LF are [control characters](https://en.wikipedia.org/wiki/Control_character) or [bytecode](https://en.wikipedia.org/wiki/Bytecode) that can be used to mark a line break in a text file.
>
> CR = Carriage Return (`\r`, `0x0D` in hexadecimal, `13` in decimal) — moves the cursor to the beginning of the line without advancing to the next line.
> LF = Line Feed (`\n`, `0x0A` in hexadecimal, `10` in decimal) — moves the cursor down to the next line without returning to the beginning of the line.

To render the progress bar, `avrdude` repeatedly overwrites the same line using `CR` as a control character to move the cursor back at the beginning of the line, without advancing to the next one. The issue here is that the IDE text area interprets `CR` as `CR+LF` moving the cursor to the next line.

IMHO this issue should be moved to the Arduino IDE repo.

cmaglie, 2022-09-20:
Another question is: isn't the "console" text area already supporting some terminal control standard?

If we must implement all control characters from an ANSI terminal it will become a project on its own... maybe there is already an implementation of a "terminal" or a "console" text area that support ANSI (or vt100 or any other terminal standard) that can be reused.