arduino/arduino-cli #1885: Port on uploding

VincentR1, 2022-09-20:
### Describe the problem

The sketch is uploaded to the wrong board on dieffrent port then specifyed

### To reproduce

I have 2 simulare boards 
the first on port  /dev/ttyACM1
the second one is on port /dev/ttyACM2

I tryed to upload my sketch  to the 2. board using: `arduino-cli upload -p /dev/ttyACM2 -b Infineon:arm:XMC1100_XMC2GO`
It's run succesfully but the sketch gets uploaded to the 1. board, the 2. board, i targeted, is uneffected
 

### Expected behavior

the second board should be updated

### Arduino CLI version

nightly-20220920

### Operating system

Linux

### Operating system version

debian bullseye

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the problem still occurs when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2022-09-21:
Could you provide a copy of the verbose output of the upload (with the `-v` flag)?
Do you know if the same setup working as expected with Arduino IDE 1.8.x?

VincentR1, 2022-09-27:
I tried it with the ide 1.8.19.
 It updated the sensor on on the first port regardless wich port i choosed in "Tools->Ports"

VincentR1, 2022-09-27:
arduino-cli upload -v -p /dev/ttyACM4 -b Infineon:arm:XMC1100_XMC2GO
java -jar "/home/ /.arduino15/packages/Infineon/tools/XMCFlasher/1.2.1/XMCFlasher.jar" -p "/tmp/arduino-sketch-47D7AF5586EB509566B17F69DFB282A0/HallSensor.ino.hex"  -device "XMC1100-0064"
--------------------------
Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Linux
Operating System: Linux

Initialisation
--------------------------
Linux
Loading JLink Library... 
JLink Library Version: V7.66b

Configuring Device
Device Name : XMC1100-0064
Interface   : SWD
Speed       : 4000
File Upload
--------------------------
Uploading /tmp/arduino-sketch-47D7AF5586EB509566B17F69DFB282A0/HallSensor.ino.hex
... Done
--------------------------
... Finished Succesfully

VincentR1, 2022-09-27:
```arduino-cli upload -v -p /dev/ttyACM4 -b Infineon:arm:XMC1100_XMC2GO
java -jar "/home/v/.arduino15/packages/Infineon/tools/XMCFlasher/1.2.1/XMCFlasher.jar" -p "/tmp/arduino-sketch-47D7AF5586EB509566B17F69DFB282A0/HallSensor.ino.hex"  -device "XMC1100-0064"
--------------------------
Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Linux
Operating System: Linux

Initialisation
--------------------------
Linux
Loading JLink Library... 
JLink Library Version: V7.66b

Configuring Device
Device Name : XMC1100-0064
Interface   : SWD
Speed       : 4000
File Upload
--------------------------
Uploading /tmp/arduino-sketch-47D7AF5586EB509566B17F69DFB282A0/HallSensor.ino.hex
... Done
--------------------------
... Finished Succesfully
```

cmaglie, 2022-09-28:
In your board's platform, the serial port is not used to select the upload target (I guess they use it only for the serial monitor), you can see it because the port is not used at all in the command line:

```
java -jar "/home/v/.arduino15/packages/Infineon/tools/XMCFlasher/1.2.1/XMCFlasher.jar" -p "/tmp/arduino-sketch-47D7AF5586EB509566B17F69DFB282A0/HallSensor.ino.hex"  -device "XMC1100-0064"
```

It seems that they are using the SWD protocol to upload that "autodetect" the board to upload to. In this case, it is just uploading to the first available board.
I don't think there is much we can do from the Arduino-CLI side.

per1234, 2022-09-28:
We are tracking the request for a capability to specify which board to target under these conditions here: https://github.com/arduino/arduino-cli/issues/770

per1234, 2022-09-28:
Actually, thinking about it more, this is already available for uploads (but not for `burn-bootloader` as requested at https://github.com/arduino/arduino-cli/issues/770):

https://arduino.github.io/arduino-cli/dev/platform-specification/#user-provided-fields

closed 2022-09-28 by per1234