arduino/arduino-cli #1892: Compilation failure if directory too long

MooMooMod, 2022-09-24:
Spent about an hour trying to understand why moving a portable arduino install compiles in some folders but not others.
To recreate - keep putting install in more and more folders until the basic file, with no code inside setup() or loop() procedures, has a compilation failure.

ERROR LOG:
```
In file included from C:\Users\gamin\Desktop\Programming All 1\stm32\akjdhfjkldshfdafgaastmlink_template\portable\packages\stm32duino\hardware\STM32F1\2021.5.31\cores\maple\cxxabi-compat.cpp:1:0:
c:\users\gamin\desktop\programming all 1\stm32\akjdhfjkldshfdafgaastmlink_template\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\cstdlib:41:28: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
                            ^
compilation terminated.
exit status 1
Error compiling for board Generic STM32F103C series.
```

Changing name from **akjdhfjkldshfdafgaastmlink_template** to **aastmlink_template** compiles just fine.

cmaglie, 2023-04-27:
Unfortunately, this is a combination of two problems of `gcc` and Windows:
* Windows limits the path to ~`256` chars (more or less I don't remember the exact number :-))
* `gcc` generates some include search path with a lot of `../../../..` path traversal reducing the available chars even more.

I'll keep this issue open to track the problem but it doesn't have an easy solution AFAIK.

Puiching-Memory, 2025-05-10:
> Unfortunately, this is a combination of two problems of and Windows:`gcc`
> 
> * Windows limits the path to ~ chars (more or less I don't remember the exact number :-))`256`
> * `gcc` generates some include search path with a lot of  path traversal reducing the available chars even more.`../../../..`
> 
> I'll keep this issue open to track the problem but it doesn't have an easy solution AFAIK.

Hello, has this problem been solved now?