arduino/arduino-cli #1894: Add `author` field to library search scope

per1234, 2022-09-24:
### Describe the request

Add the `author` field of the library index to the search scope of the `arduino-cli lib search` command.

🙂 The chances of finding a library by searching for the name of its author will be increased.

```text
$ arduino-cli lib search "cristian maglie"
Downloading index: library_index.json.gz downloaded
Downloading index signature: library_index.json.sig downloaded
Name: "Arduino_ConnectionHandler"
  Author: Ubi de Feo, Cristian Maglie, Andrea Catozzi, Alexander Entinger et al.
  Maintainer: Arduino.cc
  Sentence: Arduino Library for network connection management (WiFi, GSM, NB, [Ethernet])
  Paragraph: Originally part of ArduinoIoTCloud
  Website: https://github.com/arduino-libraries/Arduino_ConnectionHandler
  Category: Communication
  Architecture: samd, esp32, esp8266, mbed, megaavr, mbed_nano, mbed_portenta, mbed_nicla
  Types: Arduino
  Versions: [0.1.0, 0.1.2, 0.1.3, 0.1.4, 0.2.0, 0.3.1, 0.3.2, 0.3.3, 0.4.0, 0.4.1, 0.4.2, 0.4.3, 0.4.4, 0.4.5, 0.4.6, 0.4.7, 0.4.8, 0.4.9, 0.5.0, 0.5.1, 0.6.0, 0.6.1, 0.6.2, 0.6.3, 0.6.4, 0.6.5, 0.6.6]
  Dependencies: Arduino_DebugUtils, WiFi101, WiFiNINA, MKRGSM, MKRNB, MKRWAN
Name: "FlashStorage_SAMD"
  Author: Cristian Maglie <c.maglie@bug.st>, Khoi Hoang <khoih.prog@gmail.com>
  Maintainer: Khoi Hoang <khoih.prog@gmail.com>
  Sentence: The FlashStorage library aims to provide a convenient way to store and retrieve user's data using the non-volatile flash memory of SAMD21/SAMD51. It's using the buffered read and write to minimize the 
access to Flash. It now supports writing and reading the whole object, not just byte-and-byte.
  Paragraph: Useful if the EEPROM is not available or too small. Currently, ATSAMD21 and ATSAMD51 are supported (and consequently every board based on this cpu like the Arduino Zero, Aduino MKR1000, Nano-33 IoT, Itsy-Bitsy M4, etc).
  Website: https://github.com/khoih-prog/FlashStorage_SAMD
  License: GPLv3
  Category: Data Storage
  Architecture: samd
  Types: Contributed
  Versions: [1.0.0, 1.1.0, 1.2.0, 1.2.1, 1.3.0, 1.3.1, 1.3.2]
  Provides includes: FlashStorage_SAMD.h, FlashStorage_SAMD.hpp, FlashAsEEPROM_SAMD.h, FlashAsEEPROM_SAMD.hpp
```

### Describe the current behavior

The scope of `arduino-cli lib search` is the following fields of [the library index](http://downloads.arduino.cc/libraries/library_index.json):

https://github.com/arduino/arduino-cli/blob/5730e2ebbed39452a3a0517bb3571a1b94d864f7/commands/lib/search.go#L44

- `name`
- `sentence`
- `paragraph`

The `author` field of [the `library.properties` metadata file of Arduino libraries](https://arduino.github.io/arduino-cli/dev/library-specification/#libraryproperties-file-format) contains the name of the author(s) of the library.

🙁 It is difficult to find libraries by searching for the author name.

```text
$ arduino-cli version
arduino-cli.exe  Version: git-snapshot Commit: 5730e2eb Date: 2022-09-24T17:05:10Z

$ arduino-cli lib search "cristian maglie"
Downloading index: library_index.json.gz downloaded
Downloading index signature: library_index.json.sig downloaded
No libraries matching your search.
```

### Arduino CLI version

5730e2eb

### Operating system

Windows

### Operating system version

10

### Additional context

Originally requested by @tablatronix at https://github.com/arduino/Arduino/issues/11679

---

Related: https://github.com/arduino/arduino-cli/issues/1893

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

tablatronix, 2022-09-25:
Thanks for reposting for me, it seems the library manager also only shows one author which can also be problematic. So fixing both the display in the ide and helping with search inside the ide lib manager

per1234, 2022-09-25:
Hi @tablatronix
 
>it seems the library manager also only shows one author which can also be problematic

I think you are referring to a bug specific to Arduino IDE 1.x, as reported in your issue as well as others:

- https://github.com/arduino/Arduino/issues/11679
- https://github.com/arduino/Arduino/issues/11618
- https://github.com/arduino/Arduino/issues/3329

However, I can not reproduce this in Arduino CLI and Arduino IDE 2.x:

```text
$ arduino-cli lib search FlashStorage_SAMD

[...]

Name: "FlashStorage_SAMD"
  Author: Cristian Maglie <c.maglie@bug.st>, Khoi Hoang <khoih.prog@gmail.com>
  Maintainer: Khoi Hoang <khoih.prog@gmail.com>
  Sentence: The FlashStorage library aims to provide a convenient way to store and retrieve user's data using the non-volatile flash memory of SAMD21/SAMD51. It's using the buffered read and write to minimize the 
access to Flash. It now supports writing and reading the whole object, not just byte-and-byte.
  Paragraph: Useful if the EEPROM is not available or too small. Currently, ATSAMD21 and ATSAMD51 are supported (and consequently every board based on this cpu like the Arduino Zero, Aduino MKR1000, Nano-33 IoT, Itsy-Bitsy M4, etc).
  Website: https://github.com/khoih-prog/FlashStorage_SAMD
  License: GPLv3
  Category: Data Storage
  Architecture: samd
  Types: Contributed
  Versions: [1.0.0, 1.1.0, 1.2.0, 1.2.1, 1.3.0, 1.3.1, 1.3.2]
  Provides includes: FlashStorage_SAMD.h, FlashStorage_SAMD.hpp, FlashAsEEPROM_SAMD.h, FlashAsEEPROM_SAMD.hpp
```

[image]

If you are able to reproduce it with either Arduino CLI or Arduino IDE 2.x, please do submit a bug report about that.

tablatronix, 2022-09-25:
I did not know about ide 2.x ill test it

cmaglie, 2022-09-28:
Fixed by https://github.com/arduino/arduino-cli/pull/1896

closed 2022-09-28 by cmaglie