arduino/arduino-cli #1917: A way to specify "macOS universal" tool host in index.json

carlosperate, 2022-10-11:
### Describe the request

When building an Arduino Core and packaging tools, it looks like there isn't a way to specify a macOS universal "host".

Having a look in https://downloads.arduino.cc/packages/package_index.json I'm not 100% sure what's the right name for the arm64 macOS host, as I can't find any entry there. But based on this test file I believe it's `arm64-apple-darwin`?:
https://github.com/arduino/arduino-cli/blob/495a5f1a3d9d2f47b399329d6e6d91562fb683f8/arduino/cores/tools_test.go#L67

### Describe the current behavior

We can have an entry for `arm64-apple-darwin` and another for `x86_64-apple-darwin`, which would be pretty much a copy/paste of the other with a different "host" name.

### Arduino CLI version

N/A

### Operating system

macOS

### Operating system version

N/A

### Additional context

This is not extrictly neccesary, as having multiple entries works, so I guess it's more like a "nice to have".

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-10-12:
Hi @carlosperate. You can see the code for selecting the appropriate tool flavor here:

https://github.com/arduino/arduino-cli/blob/08d746dd9c31ce75b94a2c3caced3e34b6687f6a/arduino/cores/tools.go#L126-L213

So a host value matching the regular expression `i[3456]86-apple-darwin.*` will cause the tool to be used on:

- macOS x86
- macOS x86-64
- macOS ARM

A host value matching the regular expression `x86_64-apple-darwin.*` will cause the tool to be used on:

- macOS x86-64
- macOS ARM

I aspired to document the supported values of `packages[*].tools[*].systems[*].host` at one time, but never managed to figure out how to do it.

cmaglie, 2023-04-28:
AFAIU, this is a documentation problem? can we close this in favor of #2004?

closed 2023-05-16 by carlosperate