arduino/arduino-cli #1923: Allow installing from GIT at specific version

FStefanni, 2022-10-14:
### Describe the request

Hi,

currently the `arduino-cli lib install --git-url` command does not allow to specify a version.
So it could be useful to allow this, by supporting tag, branches and SHA ID's.

Git already support this with the `--branch` option.

So something as this should be fine:
```
arduino-cli lib install --git-url https://www.super.url/repo@1.2.3
arduino-cli lib install --git-url https://www.super.url/repo@abcde
arduino-cli lib install --git-url https://www.super.url/repo@featurex-branch
```

Regards

### Describe the current behavior

Currently always the latest commit on master/main is downloaded.

### Arduino CLI version

0.27.1

### Operating system

Linux

### Operating system version

Debian testing

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-cli/issues?q=)
- [X] I verified the feature was still missing when using the [nightly build](https://arduino.github.io/arduino-cli/dev/installation/#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-10-14:
Hi @FStefanni. Thanks for your request. I'm happy to tell you this capability has already been added thanks to a community contribution: https://github.com/arduino/arduino-cli/pull/1776

closed 2022-10-14 by per1234

FStefanni, 2022-10-14:
Hi,

that's great.
Sorry, I tried with '@' as for normal libs, without success, and found no description in the doc.
So maybe this could be mentioned in the doc: for urls, use `#`.

Regards.

per1234, 2022-10-14:
It is demonstrated in the examples provided on the documentation website:

https://arduino.github.io/arduino-cli/dev/commands/arduino-cli_lib_install/#examples

And in the command line help:

```text
$ ./arduino-cli lib install --help
Installs one or more specified libraries into the system.

Usage:
  arduino-cli lib install LIBRARY[@VERSION_NUMBER]... [flags]

Examples:
  arduino-cli lib install AudioZero       # for the latest version.
  arduino-cli lib install AudioZero@1.0.0 # for the specific version.
  arduino-cli lib install --git-url https://github.com/arduino-libraries/WiFi101.git https://github.com/arduino-libraries/ArduinoBLE.git
  arduino-cli lib install --git-url https://github.com/arduino-libraries/WiFi101.git#0.16.0 # for the specific version.
```

Any suggestions for improvements on the documentation are welcome.