arduino/arduino-cli #1925: Remove board attach command and related gRPC interface function

umbynos, 2022-10-17:
The `board attach` command has a broken implementation.
It won't allow passing the correct flags (`-b` / `-p`) but by default it will only accept an `fqbn` or a port when the latter is passed in as a full path (e.g.:  `serial:///dev/cu.usbmodem14301`)

Passing both parameters at once leads to an error

```
$ arduino-cli board attach arduino:samd:mkr1000
Selected fqbn: arduino:samd:mkr1000

$ cat sketch.json
{
  "cpu": {
  "fqbn": "arduino:samd:mkr1000",
  "port": ""
  }
}
```

Passing in the serial port will resolve to an `fqbn` and fill the `sketch.json` file with all the required information
```
$ arduino-cli board attach serial:///dev/cu.usbmodem614601
Board found: Arduino MKR1000...
Selected fqbn: arduino:samd:mkr1000

$ cat sketch.json
{
  "cpu": {
  "fqbn": "arduino:samd:mkr1000",
  "name": "Arduino MKR1000",
  "port": "serial:///dev/cu.usbmodem614601"
  }
}
```

Once the `sketch.json` file is filled, any `compile`/`upload` operation won't require the `fqbn` or `port` parameters.

e.g.:
```
$ arduino-cli compile -v
Using board 'mkr1000' from platform in folder: /Users/ubi/Library/Arduino15/packages/arduino/hardware/samd/1.8.11
[...]
Sketch uses 10996 bytes (4%) of program storage space. Maximum is 262144 bytes.
Global variables use 1992 bytes (6%) of dynamic memory, leaving 30776 bytes for local variables. Maximum is 32768 bytes.
```

``` 
$ arduino-cli upload
Atmel SMART device 0x10010005 found
Device       : ATSAMD21G18A
Chip ID      : 10010005
Version      : v2.0 [Arduino:XYZ] Apr 11 2019 13:09:51
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : true
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
done in 0.819 seconds

Write 10996 bytes to flash (172 pages)
[==============================] 100% (172/172 pages)
done in 0.072 seconds

Verify 10996 bytes of flash with checksum.
Verify successful
done in 0.010 seconds
CPU reset.
```

All in all I still believe this command has some potential, but not having proper flags set for it has hindered its usage customs and appears to be a leftover from times past

per1234, 2022-10-17:
The deficiency of being unable to specify both port and FQBN via the command is tracked at https://github.com/arduino/arduino-cli/issues/1073

ubidefeo, 2022-10-18:
@per1234 
this issue was only tracked internally. 
Yesterday, during grooming/planning, we made it public in order to add it to the `1.0` milestone.
This is why this one popped up seemingly out of the blue.
Your mention of #1073 publicly links the two 🙏🏼

silvanocerza, 2022-10-19:
Am doing this.

umbynos, 2022-10-19:
:metal:

PR #1930 (2022-11-21): [breaking] Fix board attach CLI command, and remove gRPC API
## Please check if the PR fulfills these requirements

See [how to contribute](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/)

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes  [... cut]
closed 2022-11-21 by cmaglie